jneilliii/OctoPrint-BedLevelVisualizer #485: [BUG]: 1.1.1rc4 crashes on display of mesh

AlAmantea, 2021-11-09:
**Describe the bug**

After updating to 1.1.1rc4, Bed Visualizer reports a crash, stating that Plotly is not defined.

This has occurred with 4 different setups.

**Expected behavior**

Bed Mesh should be displayed, but nothing is shown on screen.

**Debug Log**

[2021-11-08 18:28:21,508] DEBUG: mesh collection started
[2021-11-08 18:28:46,636] DEBUG: mesh collection started
[2021-11-08 18:29:38,400] DEBUG: ['-0.032', '-0.032', '-0.032', '-0.005', '+0.075']
[2021-11-08 18:29:38,403] DEBUG: ['+0.010', '-0.015', '-0.040', '-0.055', '-0.005']
[2021-11-08 18:29:38,404] DEBUG: ['+0.083', '+0.043', '+0.000', '-0.047', '-0.055']
[2021-11-08 18:29:38,406] DEBUG: ['+0.035', '+0.000', '-0.032', '-0.070', '-0.075']
[2021-11-08 18:29:38,408] DEBUG: ['-0.155', '-0.155', '-0.130', '-0.115', '-0.082']
[2021-11-08 18:29:38,607] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 300.0, 'y_min': 0, 'y_max': 300.0, 'z_min': 0, 'z_max': 400.0}
[2021-11-08 18:29:38,609] DEBUG: stopping mesh collection
[2021-11-08 18:29:38,610] DEBUG: Final mesh:
[2021-11-08 18:29:38,610] DEBUG: ['-0.155', '-0.155', '-0.130', '-0.115', '-0.082']
[2021-11-08 18:29:38,611] DEBUG: ['+0.035', '+0.000', '-0.032', '-0.070', '-0.075']
[2021-11-08 18:29:38,611] DEBUG: ['+0.083', '+0.043', '+0.000', '-0.047', '-0.055']
[2021-11-08 18:29:38,611] DEBUG: ['+0.010', '-0.015', '-0.040', '-0.055', '-0.005']
[2021-11-08 18:29:38,612] DEBUG: ['-0.032', '-0.032', '-0.032', '-0.005', '+0.075']

**Screenshots**

[image]

**Firmware and Version**

Marlin 2.0.3.x

jneilliii, 2021-11-09:
Just released 1.1.1rc5 with the missing file, thanks for the report.

closed 2021-11-09 by jneilliii

AlAmantea, 2021-11-09:
1.1.1.rc5 is not showing on octoprint software update

Al Amantea
Amantea Fine Woodworks
12455 S Harrell's Ferry Rd
Baton Rouge, LA 70816
225.571.1151 cell
***@***.***

jneilliii, 2021-11-10:
I just released it again, you may have to use the advanced options in software update and force check for updates.

AlAmantea, 2021-11-10:
Got it done. Thanks! Working perfectly on all 5 setups now. Al From: jneilliiiSent: Tuesday, November 9, 2021 11:50 PMTo: jneilliii/OctoPrint-BedLevelVisualizerCc: AlAmantea; AuthorSubject: Re: [jneilliii/OctoPrint-BedLevelVisualizer] [BUG]: 1.1.1rc4 crashes on display of mesh (Issue #485) I just released it again, you may have to use the advanced options in software update and force check for updates.