jneilliii/OctoPrint-BedLevelVisualizer #490: [BUG]: Decending X-Axis changes axis titles and data, but not tooltips/hover

jberry, 2021-12-14:
**Describe the bug**
When you select "Descending x-axis", the data and axis titles rotate, but when you hover-over the data, it's showing the non-flipped-data for rotation

**Expected behavior**
I expect the tooltips/hover to be correct

**Screenshots**
[image]
[image]

**Firmware and Version**
Bed Visualizer Settings version 1.1.0

jneilliii, 2021-12-14:
Thank you for your report. When I get some time I'll investigate this further. I've never really been happy with how this table works (was contributed by someone else).

jneilliii, 2022-01-01:
rather than actually fixing this I'm just going to take the hover tooltip out completely. It just doesn't seem to be beneficial and taking it out should actually improve processing time.

jberry, 2022-01-03:
This did not resolve the data issue of the right chart, the data is still flipped:
[image]

jneilliii, 2022-01-03:
Sorry I wasn't paying attention. I thought the issue that the arrow on the mouse pop-over was incorrect, not the actual calculated turns.

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]