jneilliii/OctoPrint-BedLevelVisualizer #492: [FR]:

Gwebber-Net, 2021-12-21:
**Is your feature request related to a problem? Please describe.**

Not being able to see if my printer is actually doing bed levelling.

**Describe the solution you'd like**

It would be nice, if i could see the camera feed while doing this process.
So i know the printer is actually doing something
**Describe alternatives you've considered**

Running back and forth from my pc to the printer
**Additional context**

Thegs68, 2021-12-22:
Hi
There is an option in Octoprint/settings/plugins/Bed Visualizer/Collection tab, there you can find the option to have the “Webcam while processing” at the bottom.
Hope this helps.

jneilliii, 2021-12-22:
Thanks @Thegs68, that is correct.

closed 2021-12-22 by jneilliii