jneilliii/OctoPrint-BedLevelVisualizer #493: [BUG]:

K20FRITZ, 2021-12-23:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

jneilliii, 2021-12-23:
could you enable debug logging in the plugin's settings and attempt to update mesh again? Then share the octoprint.log, plugin_bedlevelvisualizer_debug.log and check your browser's developer tools, console tab for any errors on initial page load. I think you may be running into some form of conflict or loading issue. I run the plugin in multiple instances running python 3 and OctoPrint 1.7.2.

K20FRITZ, 2021-12-23:
[image]

K20FRITZ, 2021-12-23:
[image]

K20FRITZ, 2021-12-23:
[image]

jneilliii, 2021-12-23:
you can upload those files here by dragging them into the comment. Nothing wrong in those. What multicam plugin are you using because I believe that is what's interfering here.

K20FRITZ, 2021-12-23:
[octoprint-logs.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7771247/octoprint-logs.zip)

jneilliii, 2021-12-23:
yeah, the data is getting collected so this is definitely a front-end conflict. What is the multi-cam plugin that you are using for displaying multiple cameras as that seems to be what's showing up here?

K20FRITZ, 2021-12-23:
its not the cam its the actual plotter. but to test ive removed multicam and re tested

K20FRITZ, 2021-12-23:
[image]

closed 2021-12-23 by K20FRITZ