jneilliii/OctoPrint-BedLevelVisualizer #494: [FR]: Snapshots of Rendered Graph - Name and Location

rachmo, 2021-12-30:
**Is your feature request related to a problem? Please describe.**
Downloaded renders do not have descriptive enough titles and are spread across multiple computers-instead of being tied to specific printer instances.

**Describe the solution you'd like**
1. The file name for the rendered download should have the localhost and/or printer name (appearance > "title" is also an option) appended to the date so renders from different printers can be distinguished
2. Ideally the rendered graph has the option to be saved to the raspberry pi instead of downloading. This allows the data to stay with the printer (like a time-lapse) 

**Describe alternatives you've considered**
If it's not reasonable to save the renders to the raspberry pi, at a minimum appending the local host name or the "appearance" > "title" to the file name would allow the different printers to be identified.

**Additional context**
I have multiple printers (10+) and right now when the images download they all just get mixed up (no distinguishable file names) and when I use different computers to access my printers the files get spread across multiple locations as well. These printers are used by many people and I would like to regularly go through and do a quick assessment of which ones need attention to the bed level.

Thanks for your time!

jneilliii, 2021-12-30:
Thanks, that's a good idea. Unfortunately, option 2 won't be available as the graphs are rendered in the browser of the client, not server side. I would lean towards using the appearance > title instead of hostname as that is more easily changeable by the end user. 

Relative to #2 though, I did create another plugin that you may be interested in...https://github.com/jneilliii/OctoPrint-BLVLogger. It actually stores the mesh and then allows you to switch the graph in OctoPrint itself.

rachmo, 2021-12-31:
Oh that make sense on the limitation. And thanks for pointing me to the other plugin you developed, that seems like what I needed. Thanks!!

I think appending the appearance > title to the file name would still be a nice improvement, but feel free to close this if it's not something you want to pursue.

jneilliii, 2022-01-01:
added with the above commit and will be available in version 1.1.1rc6 or the plugin shortly. You can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and upgrade when prompted to test and validate.

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]
closed 2022-01-01 by jneilliii