jneilliii/OctoPrint-BedLevelVisualizer #495: [FR]: Add options for X-Axis, Y-Axis, and background CSS color

ceshop-nick, 2021-12-31:
Did not notice there were colors for X and Y axis labels until yesterday.  The black and green colors are hard to distinguish on my screen.

Add input boxes on settings tab for x-axis, y-axis, and background colors

I really like the plugin but would have been so much more useful being able to distinguish axis better.

jneilliii, 2021-12-31:
Curious if you've tried the release candidate version of the plugin? If I remember correctly I have added some better theme support for readability. If not please share a screenshot. Totally open to making it configurable, but wanted to get an idea if my automated approach might work.

ceshop-nick, 2022-01-01:
I'm using the latest version of it.  Where is the theme support you are mentioning?  If it's already in there to use via another method, that's fine.  I just wanted to use a more contrasted color between labels for myself.  I see where it is in the css code and changed the green to a goldenrod color (probably not the prettiest, but I could see the difference better) using chrome inspector.

jneilliii, 2022-01-01:
In Software Update settings change the Release Channel for the plugin to `Release Candidate`.

jneilliii, 2022-01-30:
Just following up here on this one to see if the latest 1.1.1 version of the plugin satisfies Theme related issues?

closed 2022-07-16 by jneilliii