jneilliii/OctoPrint-BedLevelVisualizer #496: [FR]: Option to display height instead of angles on main screen.

ceshop-nick, 2021-12-31:
Option to display mesh data height instead of only angles on main screen.

Display Mesh Data on tab screen with checkbox same as angle correction.  Then user could have both if they wanted.

jneilliii, 2021-12-31:
Nice idea. I never thought to do that since you can hover your mouse over the points on the graph and it shows the height. I can see now seeing all the numbers can be helpful in some situations, like on your phone.

jneilliii, 2022-01-01:
Need to work out the math changes needed on the adjustment area, as that doesn't seem to showing any numbers. It may take a while for me to figure that out though because I didn't create this part of the plugin and it will take me a while to understand how the contributor did what they did. 

[image]

ceshop-nick, 2022-01-01:
Thank you for being so proactive on this on New Years eve(n).  What if you just copied the code to display on the main page from the angles to the Mesh Data page since that is already displaying the mesh data in mm already?  Then you could potentially have 3 sections on the page - Graph, Angles View, Mesh Data view - (up to the user to display all three or not).

jneilliii, 2022-01-01:
That is actually a screenshot of the tab, just below the graph, so all three items can be enabled and shown on the tab all at the same time if configured that way.

PR #497 (2022-01-01): 1.1.1
* add ability to configure graph height in visualization settings, resolves #462
* add min, max, and variance values of graphed mesh, resolves #286
* add OctoDash support, create Custom Action with `[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer` as the command.
* adjust the tables for better theme support, resolves #479
* add option for showing mesh data on tab, resolves #496
* remove tooltip hover on tables, resolves #490
* add Prusa adjustment values to graph (still needs verification), resolves #242
* add appearance > title setting to automatically downloade [... cut]
closed 2022-01-01 by jneilliii