jneilliii/OctoPrint-BedLevelVisualizer #498: [FR]: Automatically obtain bed values from pre-print bed level.

TheConsciousness, 2022-01-04:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
Instead of having to run 'Update Mesh Now' to get the latest mesh, I'd like to have the mesh automatically updated once my pre-print G29 command completes its leveling process and the results are returned.

**Describe alternatives you've considered**
Updating the mesh on print boot?

**Additional context**
N/A

jneilliii, 2022-01-04:
If you update your start gcode script to include `@BEDLEVELVISUALIZER` just prior to the G29 command the plugin will automatically update every time you print if the save mesh option is enabled. Is that what you mean?

TheConsciousness, 2022-01-05:
> If you update your start gcode script to include `@BEDLEVELVISUALIZER` just prior to the G29 command the plugin will automatically update every time you print if the save mesh option is enabled. Is that what you mean?

That worked perfectly! Throw that on the front page and it'll be a hit!
Thanks!

closed 2022-01-05 by TheConsciousness