jneilliii/OctoPrint-BedLevelVisualizer #499: [BUG]: Displayed "Var" value not rounded to nearest 3rd decimal

AlexVercammen, 2022-01-04:
Not really a BUG not a FEATURE, just a formatting issue.
In my case the "Var" value is 0.0399999999994. Where "Min" and "Max" values are rounded: -0.018 and 0.022
Excellent plug-in. I am definitely not using at its most potential but it gets the job done.
Snip Image included.Hopefully I've inserted it correctly.

[image]

jneilliii, 2022-01-04:
I'll take a look, the other values are probably not rounded, but returned rounded from the firmware.

jneilliii, 2022-01-30:
I've resolved this in the above commit and will be available with the next 1.1.2 release candidate for testing in the future.

closed 2022-01-30 by jneilliii