jneilliii/OctoPrint-BedLevelVisualizer #500: [FR]: Individual bed size

Alex-APK-HZ, 2022-01-07:
**Is your feature request related to a problem? Please describe.**
The dimension of the Graph and the probing points are off

**Describe the solution you'd like**
My printbed has a rectangular shape (220 mm by 195 mm). Still, the bed size of the graph is displayed as a square (220 mm by 220 mm). Furthermore the probe (BLTouch) is offset to the side, so the probe cannot reach the edge of the printbed. The probing points are spaced evenly in the accessable area. This results in the probing positions, calculated by the plugin, being off.

The best solution would be for the plugin to draw the bed size values from the active printer profile.
Furthermore it would be nice to add the possibility to edit the coordinates of each of the probing points and link the resulting grid to the selected printer profile, so the plugin automatically draws the right grid for each setup. (for example when I switch between printbeds)

**Describe alternatives you've considered**
just add a menue to set the bed size and another one to set the coordinates of each point in the grid, for example input them as a matrix.

**Additional context**
I know this project is a lot of work. Many thanks to all the developers!

jneilliii, 2022-01-07:
The plugin does utilize the printer profile to grab the dimensions of the bed. It does however, not take into account offsets from probing settings, etc. and the points are a rough estimate of where those would be based on dividing the dimensions by the number of probe points. Could you do me a favor and enable debug logging in the plugin, restart OctoPrint and then run through the update mesh process and then share your plugin_bedlevelvisualizer_debug.log file from OctoPrint's logging section?

Alex-APK-HZ, 2022-01-07:
Edited the comment, because I explained it wrong. The bed size issue only applies for the values shown in the graph, not in the actual data tab. Only the issue with the probe offset altering the probing points is present in the data tab.

I should have the file ready in about half an hour.

jneilliii, 2022-01-10:
Sorry, didn't get notice of an edit, was expecting new comment. I'll review later today.

Alex-APK-HZ, 2022-01-10:
Oh, sorry. No problem.

Alex-APK-HZ, 2022-01-11:
Yes, the available bed size is 220 x 195 (I am using an Ender 3 V2, so originally a 220 x 220, but the clips holding the bed take up some space in the y direction)
Furthermore the printhead chas the folowing movement limits:
 X min -3 
 X max 245
 Y min 0 
 Y max 225 
 Z max 235

Alex-APK-HZ, 2022-01-26:
[image]
I‘m sorry it took so long. We had a lot of trouble at work. Here is the screenshot.

jneilliii, 2022-01-26:
No problem, this is what I thought, and is very similar to other requests of this nature, #516, #434, #130.

jneilliii, 2022-01-30:
Going to close in preference of #130 for now. I believe if I can get that one it will also meet the requirements of this one.

closed 2022-01-30 by jneilliii

jneilliii, 2022-01-30:
Added the option to define probe margin/offset in above commit and is available in version 1.1.2rc2. You can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and update when prompted if you want to try it out.