jneilliii/OctoPrint-BedLevelVisualizer #502: [BUG]: No mesh visualization, just dots.

moreleftrudder, 2022-01-09:
**Describe the bug**
Mesh is not visualized, instead I see a cloud of dots.

**Expected behavior**
I expect to see a solid "sheet" that portrays the different bed heights captured.

**Debug Log**

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7834404/plugin_bedlevelvisualizer_debug.log)

**Screenshots**

[image]
[image]

**Firmware and Version**
Marlin 2.0.7.2

jneilliii, 2022-01-09:
I've never seen this happen before and is very odd to me. The plugin is definitely getting the offsets from the firmware. What browser/version are you using? Could you try setting the Z limits to `-0.3,0.3` and update mesh and see if it makes any difference?

moreleftrudder, 2022-01-10:
The problem occurs on both Safari 15.1 and Chrome 97.0.4692.71.  Attached is the image from an updated mesh with Z limits set to -0.3, 0.3
[image]
.

jneilliii, 2022-01-10:
That is so strange. I'll have to do some checking on the upstream plotly js library to see if I can find out anything about this. My best guess is that it has something to do with either resolution or webgl/driver issues.

jneilliii, 2022-01-10:
quick question, are there any errors in developer tools/console tab of your browser on initial page load, or when the graph is being rendered?

moreleftrudder, 2022-01-13:
> quick question, are there any errors in developer tools/console tab of your browser on initial page load, or when the graph is being rendered?

Oddly, Chrome started working.  I didn't change or update anything.  Safari is still all dots.

Here's the console output for Safari when reloading the bed visualizer tab:

[Error] Unhandled Promise Rejection: TypeError: undefined is not an object (evaluating 'i.generateKey')
	promiseEmptyOnRejected (content-script.js:75:780)
	promiseReactionJob
[Info] Starting dependency resolution... (packed_core.js, line 1244)
[Info] ... dependency resolution done (packed_core.js, line 1257)
[Info] Initial application setup done, connecting to server... (packed_core.js, line 1280)
[Info] Connected to the server (packed_core.js, line 1081)
[Info] Triggering connect on all view models (packed_core.js, line 1087)
[Info] Server (re)connect processed (packed_core.js, line 1089)
[Info] Finalizing application startup (packed_core.js, line 1280)
[Info] Going to bind 44 view models... (packed_core.js, line 1261)
[Log] octolapse.help.js - Binding help links to #octolapse_tab a.octolapse_help[data-help-url] (packed_plugins.js, line 17150)
[Info] Did not bind view model – "TimelapseViewModel" – "to target" – "#timelapse" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "TimelapseViewModel" – "to target" – "#timelapse_link" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "BackupViewModel" – "to target" – "#wizard_plugin_backup" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "UsageViewModel" – "to target" – "#wizard_plugin_tracking" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "SoftwareUpdateViewModel" – "to target" – "#softwareupdate_confirmation_dialog" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "SoftwareUpdateViewModel" – "to target" – "#wizard_plugin_softwareupdate_update" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "SoftwareUpdateViewModel" – "to target" – "#wizard_plugin_softwareupdate_settings" – "since it does not exist" (packed_core.js, line 1273)
[Info] Did not bind view model – "bedlevelvisualizerViewModel" – "to target" – "#wizard_plugin_bedlevelvisualizer" – "since it does not exist" (packed_core.js, line 1273)
[Info] User tom logged in (packed_core.js, line 216)
[Info] ... binding done (packed_core.js, line 1278)
[Info] Application startup complete (packed_core.js, line 1278)
[Warning] The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-brands-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-regular-400.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff2?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Warning] The resource http://octopi.local/static/vendor/font-awesome-5.15.1/webfonts/fa-solid-900.woff?v=5.15.1 was preloaded using link preload but not used within a few seconds from the window's load event. Please make sure it wasn't preloaded for nothing.
[Error] Stream Error, but src is empty.
	(anonymous function) (packed_plugins.js:7669)
	dispatch (packed_libs.js:13:43096)
[Log] octolapse.help.js - Binding help links to .octolapse .webcam_settings a.octolapse_help[data-help-url] (packed_plugins.js, line 17150, x2)

jneilliii, 2022-01-13:
> [Error] Stream Error, but src is empty.
(anonymous function) (packed_plugins.js:7669)

on this line, can you click that 7669 number to show the code at that point? trying to figure out if that's related to my plugin or another one.

moreleftrudder, 2022-01-13:
I think it's related to octolapse
[image]
[image]

.

jneilliii, 2022-01-13:
ok thanks, that looks like an issue with octolapse, not this plugin. just wanted to make sure.

AseKarlsson, 2022-04-11:
Hi,
The same here with Chrome.
I cannot even see dots.

//Ase
[image]

AseKarlsson, 2022-04-11:
> @AseKarlsson, you need to enable the option for relative z offset. Your points are outside the range of the z limits.

Thanks jneillill,
That fixed it fine after a new mesh calibration.
Should I do something with my printer (Anycubic Hyper) to correct this?
[image]

jneilliii, 2022-04-11:
probably no correction needed, that's just how your firmware reports the mesh.

freddo99, 2023-09-11:
I have a problem whereby I do not get any visualisation on my IPad (Safari, Chrome or Firefox). Visualisation is perfect with MacBook.

jneilliii, 2023-09-11:
I don't have an iPad to develop against, so unfortunately if it doesn't work not sure there's much I can do directly.