jneilliii/OctoPrint-BedLevelVisualizer #505: [BUG]:

nmfireman, 2022-01-09:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

nmfireman, 2022-01-09:
I have the Anycubic Vyper printer. When I run the Bed Leveling Visualizer I only get a blue line on the the X and Y axis. The graph doesn't seem to be filling in the Z axis as I don't have the picture of the entire bed.
Please let me know if you need anything else.
Thank you.

[image]

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7835509/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2022-01-09:
enable the option for using relative z offset and update the mesh again.

closed 2022-01-09 by jneilliii

nmfireman, 2022-01-09:
Thank you. I will try that after my next print finishes.

Greg.

nmfireman, 2022-01-10:
Thank you, that worked perfectly.

Greg Bouten