jneilliii/OctoPrint-BedLevelVisualizer #507: [BUG]: even after feature "added" OctoDash cannot trigger a bed level.

DigiVore-Programing, 2022-01-13:
**Describe the bug**

After you edit the OctoDash to add that option instead of leveling it does nothing.

used command: 
[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer
(by the way thats not a port that you can ping on the pi.)
**Expected behavior**

Automatic bed leveling mesh starts

**Debug Log**
[plugin_bedlevelvisualizer_debug (2).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7858838/plugin_bedlevelvisualizer_debug.2.log)

**Screenshots**
[image]

**Firmware and Version**
bugfix-2.0.x from mid 2021

jneilliii, 2022-01-13:
It works for me on my OctoDash setups. It loads the page and shows the previously saved mesh and gives a single button to update. Maybe it's because I have the autologin local plugin/config setup in my OctoPrint instances?

DigiVore-Programing, 2022-01-13:
how do you have your web settings set up?

jneilliii, 2022-01-13:
[image]

DigiVore-Programing, 2022-01-13:
yeah, that port doesnt exist on OctoPrint...

Hayden, Hayden Fixes <http://haydenfixes.jordanmccollum.com>
Repairing iPhones and computers since 2017

DigiVore-Programing, 2022-01-13:
I think if you add a plugin callout like [!BED-LEVEL] that'd work better.

jneilliii, 2022-01-13:
port 5000 does run on octoprint, but is only accessible from the device directly and I have no control over what OctoDash supports and doesn't support as it comes to the action commands. You can try without the port 5000 and just use the ip of your pi instead of 127.0.0.1, which would be required if you were using a remote octoprint that's not running on the same pi as octodash.

DigiVore-Programing, 2022-01-13:
It's on the same pi. It's the no way to add a plugin trigger? I'm be
willing to try

jneilliii, 2022-01-13:
Have you tried installing autologin plugin yet?

DigiVore-Programing, 2022-01-13:
Fairly certain that wouldn't matter, since it's OctoDash which is already
logged in.

How hard is it to make a plugin callout??

closed 2022-02-05 by github-actions[bot]