jneilliii/OctoPrint-BedLevelVisualizer #511: [FR]:  Add the Mesh meta Info to Chart Output

gpread, 2022-01-15:
**Is your feature request related to a problem? Please describe.**
For Clarity it would be nice to see (could be optional) the additional info about the current mesh from say M420 V1 (in Marlin at least)

**Describe the solution you'd like**
Either by default or selectable option (to include or exclude) show text on the chart for the following mesh info
1. Validity (Valid/Invalid)
2. Slot Identifier (Integer Number)
3. Bed Levelling Status (on/off)
4. Fade Height (off/value in mm)

jneilliii, 2022-01-15:
I find with the addition of min, max, variance, and prusa adjustment numbers to the graph output, it's already starting to get cramped. This additional info is just going to cramp it that much more. Also, some of those options are specific to UBL aren't they?

gpread, 2022-01-16:
If implemented as optional via checkbox enable/disable a la "row labels", etc. that could satisfy those who value simplicity/cleanliness over information.

From my perspective with the addition of min, max and variance, there is now an entire column area with only the bottom lines currently occupied for them.

Therefore that column area seems to have ample room to include additional data if desired.

Indeed, not all 4 of those parameters are not common to all levelling setups but neither are they unique to UBL, some are also used by other MESH or MATRIX based levelling setups.

My thought was simply that whatever additional info is returned by M420 V1 (or similar) trailing the actual Bed Level values could be parsed (and perhaps simplified for space reasons) and displayed.

Here is what I see in Octoprint Terminal as trailing info in response to M420 V1:

[image]

jneilliii, 2022-01-16:
Yeah, I'll have to ponder on the best way to process those lines, because right now I'm looking for specific regex patterns for point data, otherwise don't do anything until I see ok. I have flagged this request appropriately so the stale bot doesn't kill it and will consider this for a future release. I of course am always open to a pull request with the changes implemented.

gpread, 2022-01-16:
Thanks.

My current feeling is that its too big a learning curve for me to come up with some working code but if I find some time amongst the other things I'm working on, then I might have a play around at least trying to find regex/replace which would extract and format the values and attach here

jneilliii, 2022-01-30:
Added with above commit and available for testing in version 1.1.2rc1. Change the release channel for the plugin in OctoPrint's Software Update settings to `Release Candidate` and update when prompted. 

[image]

closed 2022-01-30 by jneilliii

gpread, 2022-01-30:
> Added with above commit and available for testing in version 1.1.2rc1. Change the release channel for the plugin in OctoPrint's Software Update settings to `Release Candidate` and update when prompted.
> 
> [image]

Just updated, enabled the feature and tested it.

It works great for me, excellent work!  Thanks for the efforts, I hope you and others will enjoy this enhancement as much as me.  👍👌😍