jneilliii/OctoPrint-BedLevelVisualizer #516: [FR]: Allow adjusting probe area bounds

Thynix, 2022-01-25:
**Is your feature request related to a problem? Please describe.**

The XY coordinates seem to assume the probe points are spread uniformly throughout the entire bed. For things like the BLTouch, or other probes where a margin from the edge is either advisable or physically required, this isn't true. This results in coordinates being incorrect. This makes it harder to add aluminum foil to compensate for deviations from flatness because the edge of the visualization isn't actually the edge of the bed, and all the coordinate numbers are wrong.

**Describe the solution you'd like**

I'd like to be able to specify the bounds of the area probed within the bed, be that explicitly or with a margin from the edge.

**Describe alternatives you've considered**

I can determine the actual probe points and consider the claimed coordinates to be their actual equivalents when making decisions.

**Additional context**

My printer sends leveling reports like

```
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.010 +0.037 +0.020 -0.029 +0.000
Recv:  1 +0.020 -0.010 -0.068 +0.006 +0.014
Recv:  2 -0.019 +0.010 -0.012 +0.020 +0.034
Recv:  3 +0.003 -0.017 -0.038 +0.079 +0.010
Recv:  4 -0.013 +0.038 -0.032 -0.014 -0.012
```

which I notice doesn't specify coordinates.

jneilliii, 2022-01-25:
You are correct, the plugin calculates x/y positional data from the number of probe points and bed size when that positional data is not reported, as you found out is rare. Adding a matching offset/margin setting would make sense to use but would "shrink" the visualization to that newly inset bed perimeter, wo would not represent the entire bed. There's no good way to counter that without loads of math calculations, so I felt it better to give the rough approximation and still see the full bed size versus clipping the bed in the visualization. If I could find a good way to do both I would, but nothing has popped up yet.

Thynix, 2022-01-27:
Ah, so Unified Bed Leveling will specify coordinates. I didn't realize that.

jneilliii, 2022-01-30:
Going to track this one under revived issue #130.

closed 2022-01-30 by jneilliii

jneilliii, 2022-01-30:
Added the option to define probe margin/offset in above commit and is available in version 1.1.2rc2. You can change the plugin's release channel in OctoPrint's Software Update settings to `Release Candidate` and update when prompted if you want to try it out.