jneilliii/OctoPrint-BedLevelVisualizer #518: [BUG]: Mesh doesn't get rendered, empty graph

xjjak, 2022-02-01:
**Describe the bug**

Running automatic bed leveling is working fine and it seems that the plugin receives the data, but the mesh doesn't get rendered, just the coordinate system.

**Expected behavior**

A color-graded "sheet". The colors would represent the captured heights.

**Debug Log**

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/7977576/plugin_bedlevelvisualizer_debug.log)

**Screenshots**
[image]
[image]
[image]
[image]

**Firmware and Version**

Marlin bugfix-2.0.x

xjjak, 2022-02-01:
Well, I think I found the issue. I finally noticed that the data points are just way too high. Trying to search for a fix now, as soon as I find one, I'll close this issue.

jneilliii, 2022-02-01:
Enable relative z offsets in the plugin's settings.

closed 2022-02-01 by jneilliii