jneilliii/OctoPrint-BedLevelVisualizer #521: [FR]: option to set number and position of levelling screws

TomW1605, 2022-02-10:
**Is your feature request related to a problem? Please describe.**

at the moment i am doing 6x6 levelling and trying to use the corrections table to fix the offsets. the issue is that the table assumes that there is a screw at every point that is probed. this almost never the case and makes it hard to know how much to actually adjust the screw by

**Describe the solution you'd like**

it would be great if we could set how many screws the bed has and what there coordinates are. then the software could use that to calculate the correct rotations for those positions based on the mesh.

in my case the screws are actually outside the bed so i understand this might not be practical as it will be outside the mesh but in many cases they are under the bed so this should not be an issue.

**Describe alternatives you've considered**

**Additional context**

jneilliii, 2022-02-10:
I think you might find the Autobim plugin helpful for your use case. It kind of automates the process using your probe and you define the screw positions.

TomW1605, 2022-02-10:
thanks i will have a look.

closed 2022-02-10 by TomW1605