jneilliii/OctoPrint-BedLevelVisualizer #523: Keeps saying wait Please wait, retrieving current mesh. It never retrieves.

nmfireman, 2022-02-18:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**

jneilliii, 2022-02-19:
Please provide the requested information in the issue template so I can better assist you.

nmfireman, 2022-02-19:
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8102574/plugin_bedlevelvisualizer_debug.log)

Not sure what else to say I think I explained it. Here is the Debug Log. I think I have your latest version. Let me know how I can  help you help me as I'm pretty new to using Octoprint and plug-ins

All I can tell you is it say's this Please wait, retrieving current mesh and never finishes. Here is a screen shot as wel
[image]
l

nmfireman, 2022-02-19:
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8102591/plugin_bedlevelvisualizer_debug.log)

nmfireman, 2022-02-19:
[image]

jneilliii, 2022-02-20:
Based on the log it appears you might not be getting a response from your printer for the commands being sent. If you run the commands one at a time in the terminal tab do you get responses for all commands? At this point I think you're sending the wrong command or you don't have Auto Bed Leveling enabled firmware.

closed 2022-03-14 by github-actions[bot]