jneilliii/OctoPrint-BedLevelVisualizer #524: [FR]:

bushibot, 2022-02-23:
Dose not seem to work correctly with Anycubic Vyper, might be due to their dev marlin version... it's unclear since that data is masked. Head moves a bit then it returns a empty table.

jneilliii, 2022-02-23:
Yeah, not much I can do if your firmware doesn't report the mesh table.

bushibot, 2022-02-23:
That’s fair, but how can I test if that is really the issue? It’s dev beach but I believe it’s pretty current build.

-Griffon

jneilliii, 2022-02-24:
Run the commands manually one at a time in the terminal tab and see how the printer responds. You may need to adjust the commands based on what the firmware expects. Assuming it's bilinear leveling you may want to try without the T on the G29 command, could be a variety of things.

closed 2022-03-18 by github-actions[bot]