jneilliii/OctoPrint-BedLevelVisualizer #529: [FR]: Instructions for Creality CR Touch

abulka, 2022-03-07:
I have a Creality Ender S1 which has [Creality CR Touch](https://www.creality3dofficial.com/products/ender-3-s1-direct-drive-3d-printer) bed levelling.

It would be helpful if the [Example GCODE](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md) page provided the relevant gcode so that I can use this plugin with my 3D printer. Presumably Creality bed levelling is common enough to provide an example for?

jneilliii, 2022-03-07:
My assumption is it will be very much similar to bilinear Marlin, but I don't have your printer/firmware to know for sure.

3ffon, 2022-03-09:
@abulka I just completed running it on my Ender 3 S1 ,

- I used the [bilinear](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md)
- flip the X axis !!!
- enjoy

closed 2022-03-31 by github-actions[bot]