jneilliii/OctoPrint-BedLevelVisualizer #530: [BUG]: Edit In Chart Studio shows empty plot because Z is not defined properly

DanielLaberge, 2022-03-08:
**Describe the bug**
The Edit In Chart Studio stopped working in recent versions. It shows an empty plot because the Z axis is not defined properly; See Screenshot.

**Expected behavior**
Z should be defined as "Z[0] Z[1] Z[2] ..." not as "y".

**Debug Log**
N/A

**Screenshots**
[image]

**Version**
Version 1.1.1

jneilliii, 2022-03-08:
I don't really control how the button works and transfers the data to chart studio, that's built into the plotly library itself. The version currently running in the plugin is fairly old though and might be the root cause. You could try my plotly 2.9.0 branch to see if it works any better for you by copying/pasting the URL below in plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/heads/test/plotly_2.9.0.zip
```

DanielLaberge, 2022-03-08:
That sounded like a good guess, but the problem persists with that branch. (Version 1.1.2rc2)

jneilliii, 2022-03-08:
Interestingly enough when I attempted this with their [example 3d surface plots](https://plotly.com/javascript/3d-surface-plots/) it doesn't load anything directly. You have to open in codepen and add the config options like this.

```
Plotly.newPlot('myDiv', data, layout, {showEditInChartStudio: true, plotlyServerURL: "https://chart-studio.plotly.com"});
```

and when I click the edit button I get this.

[image]

closed 2022-03-30 by github-actions[bot]