jneilliii/OctoPrint-BedLevelVisualizer #531: [BUG]: Printer halted due to M420

DARKFiB3R, 2022-03-08:
Bed Visualizer > Update Mesh Now > Retrieving Current Mesh...

```
Error reported by printer
Your printer's firmware reported an error. Due to that OctoPrint will disconnect. Reported error: Printer halted. kill() called!
```

Searching for a solution, I found this post... https://community.octoprint.org/t/your-printers-firmware-reported-an-error-due-to-that-octoprint-will-disconnect-reported-error-failed-to-enable-bed-leveling/24958/15

Commenting out the M420 command `M420 S1 V ; enabled leveling and report the new mesh`  in Bed Visualizer Settings > Collection, seems to fix the problem.

Previously, I had also changed this Error Handling setting in OctoPrint, https://github.com/Jyers/Marlin/wiki/OctoPrint-Settings
But I am not sure if that helped or not.

I take it that changing one or both of these settings is not usually required, but I'm not sure what is causing the issue.

Should I just leave the M420 command disabled from now on?

Thanks.

Firmware: Jyers E3V2-UBL-BLTouch-15x15-HS-v4.2.2-v2.0.1

jneilliii, 2022-03-08:
My best guess is that changing the Error Handling setting fixed the issue because I've heard of that being a problem when jyers firmware goes AWOL and doesn't report it's status back to the connected host software (OctoPrint in this instance). Whatever gcode works for your firmware that reports the mesh data to the plugin is what you should use in the settings, I'm not going to take a guess on what those gcode commands are for every possible firmware fork out there as I don't use them personally.

closed 2022-03-08 by jneilliii

DARKFiB3R, 2022-03-08:
Ah! It's giving me the error again, with no further settings changed.

It worked yesterday after disabling M420, but I adjusted my x axis hardware today, hence wanting to create a new mesh.

I understand not being able to test against all forks. Thank you for the plugin.

I guess I'll just keep poking it until it works again :)

jneilliii, 2022-03-09:
do you know if you're using UBL or Bilinear leveling? If you are using UBL there's extra steps to complete the full mesh grid using `G29 P3`, but if I remember correctly jyers defaults to bilinear. I could recommend an alternative pre-compiled Marlin fork by InsanityAutomation, which is used by TinyMachines on all the Creality printers they ship.

DARKFiB3R, 2022-03-09:
I'm using the UBL version of Jyers firmware.

Not sure I want to change firmware just yet, but I'll definitely take a look at it.

jneilliii, 2022-03-09:
you should try sending the commands manually one at a time from the terminal tab and see what happens at each stage. if you can narrow down where in the process it's breaking you can adjust accordingly. For example, I personally use UBL and my gcode settings is just `M420 V` to report the currently loaded mesh data from EEPROM. When I want to actually level I use custom buttons in the plugin to send the `G29 P1`, `G29 P3`, and `M420 S#` commands.

DARKFiB3R, 2022-03-09:
OK, I will give that a go.

Thank you for your help.

DARKFiB3R, 2022-03-09:
M420 V ended with Mesh is invalid, after showing all the mesh data.

It's 1am and my cable internet has just gone down, too. So I think I'll give it a rest for a while, before I start banging my head on the table 🤕🤣

Thanks again for your help. I will be back to annoy you some more tomorrow.... If you don't mind too much 🙂

jneilliii, 2022-03-09:
Sure. Mesh invalid is usually because it's not completed. You may need to run multiple `G29 P3` commands to fill in all the missing/unreachable probe points.

DARKFiB3R, 2022-03-09:
The 🤬 printer wasn't switched on lol.

I have it stored in a cupboard, connected to smart switch for power, and also the Pi, so I couldn't hear that it wasn't turning on fully, I could only tell that the mainboard had power because of the power supplied from the Pi.

It's happily probing away right now, but I'm not sure what settings/gcode are actually helping 🤦‍♂️