jneilliii/OctoPrint-BedLevelVisualizer #538: [FR]: what is the code for flsun SR?

platini76, 2022-04-25:
I need the code for flsun SR please

jneilliii, 2022-04-25:
I can't tell you what that is as I don't know, what command do you use to do auto bed leveling?

platini76, 2022-04-25:
I really don't know... I make autolevel from the display but before I have to mount an accessory with switch

jneilliii, 2022-04-25:
My best guess would be bilinear example in the docs. Which minimally would be the following.

```
@BEDVISUALIZER
TWO
```

platini76, 2022-04-25:
like this
[image]
?

jneilliii, 2022-04-25:
oh sorry, phone autocompleted that second line. it should be like this.

```
@BEDVISUALIZER
G29
```

jneilliii, 2022-04-25:
oh sorry, phone autocompleted that second line. it should be like this.

```
@BEDLEVELVISUALIZER
G29
```

platini76, 2022-04-25:
but I need to to it with switch mounted or without?

jneilliii, 2022-04-26:
If the firmware is configured properly it should prompt you to attach the probe, same as from leveling from the LCD.

platini76, 2022-04-26:
no it did not prompt message

jneilliii, 2022-04-26:
Did it do anything? I know my flsun q5 with Marlin flashed on it does. Not sure if stock has ABL enabled. Delta calibration and auto bed leveling are slightly different things.

platini76, 2022-04-26:
yes it worked...
thanks

[image]
I

it is good?

jneilliii, 2022-04-26:
You might want to enable relative zb offset setting.

platini76, 2022-04-26:
here
[image]
?

jneilliii, 2022-04-26:
yeah, that's the one. it will allow the reported numbers to be moved down so that the lowest point graphs at 0 instead of floating way up in the air. once enabled, just click the update mesh button again and you should see the difference.

platini76, 2022-04-26:
this is the resul
[image]
t

jneilliii, 2022-04-27:
Yep, that looks about normal for a low probe count.

platini76, 2022-04-27:
and can I also give an higher number of probe?

jneilliii, 2022-04-27:
that would require flashing new firmware, it's not controllable via gcode I'm afraid.

platini76, 2022-04-27:
so not factory firmware?

jneilliii, 2022-04-27:
correct, it would have to be custom I think. not really sure what FLSUN ships with in their speed racer.

platini76, 2022-04-27:
thank you very much.
plat