jneilliii/OctoPrint-BedLevelVisualizer #540: [BUG]: Click on "Bed Visualizer" tab and it immediately runs mesh update. Without clicking button.

z00mantwo, 2022-05-12:
**Describe the bug**
Click on "Bed Visualizer" tab and it immediately runs mesh update. Without clicking button.

**Expected behavior**
See tab without running update mesh unless I click button.

**Debug Log**
[2022-05-12 09:00:43,082] DEBUG: mesh collection started
[2022-05-12 09:02:26,845] DEBUG: mesh collection started
[2022-05-12 09:02:26,878] DEBUG: ['0.243750', '0.228156', '0.212500', '0.210234', '0.211250', '0.228406', '0.245625']
[2022-05-12 09:02:26,880] DEBUG: ['0.221516', '0.206708', '0.191625', '0.187421', '0.186125', '0.200128', '0.214406']
[2022-05-12 09:02:26,883] DEBUG: ['0.199375', '0.185563', '0.171250', '0.164984', '0.161250', '0.172250', '0.183750']
[2022-05-12 09:02:26,886] DEBUG: ['0.197813', '0.186349', '0.174234', '0.166467', '0.161219', '0.174300', '0.188031']
[2022-05-12 09:02:26,888] DEBUG: ['0.201250', '0.191625', '0.181250', '0.172531', '0.166250', '0.180250', '0.195000']
[2022-05-12 09:02:26,891] DEBUG: ['0.229109', '0.215378', '0.201188', '0.195915', '0.192000', '0.191380', '0.191219']
[2022-05-12 09:02:26,893] DEBUG: ['0.256875', '0.238828', '0.220625', '0.218922', '0.217500', '0.202109', '0.186875']
[2022-05-12 09:02:26,896] DEBUG: {'type': 'rectangular', 'x_min': 0, 'x_max': 230.0, 'y_min': 0, 'y_max': 230.0, 'z_min': 0, 'z_max': 230.0}
[2022-05-12 09:02:26,896] DEBUG: stopping mesh collection
[2022-05-12 09:02:26,897] DEBUG: flipped y axis: 
[2022-05-12 09:02:26,897] DEBUG: ['0.243750', '0.228156', '0.212500', '0.210234', '0.211250', '0.228406', '0.245625']
[2022-05-12 09:02:26,898] DEBUG: ['0.221516', '0.206708', '0.191625', '0.187421', '0.186125', '0.200128', '0.214406']
[2022-05-12 09:02:26,898] DEBUG: ['0.199375', '0.185563', '0.171250', '0.164984', '0.161250', '0.172250', '0.183750']
[2022-05-12 09:02:26,898] DEBUG: ['0.197813', '0.186349', '0.174234', '0.166467', '0.161219', '0.174300', '0.188031']
[2022-05-12 09:02:26,899] DEBUG: ['0.201250', '0.191625', '0.181250', '0.172531', '0.166250', '0.180250', '0.195000']
[2022-05-12 09:02:26,899] DEBUG: ['0.229109', '0.215378', '0.201188', '0.195915', '0.192000', '0.191380', '0.191219']
[2022-05-12 09:02:26,899] DEBUG: ['0.256875', '0.238828', '0.220625', '0.218922', '0.217500', '0.202109', '0.186875']
[2022-05-12 09:02:26,900] DEBUG: Final mesh:
[2022-05-12 09:02:26,900] DEBUG: ['0.243750', '0.228156', '0.212500', '0.210234', '0.211250', '0.228406', '0.245625']
[2022-05-12 09:02:26,901] DEBUG: ['0.221516', '0.206708', '0.191625', '0.187421', '0.186125', '0.200128', '0.214406']
[2022-05-12 09:02:26,901] DEBUG: ['0.199375', '0.185563', '0.171250', '0.164984', '0.161250', '0.172250', '0.183750']
[2022-05-12 09:02:26,901] DEBUG: ['0.197813', '0.186349', '0.174234', '0.166467', '0.161219', '0.174300', '0.188031']
[2022-05-12 09:02:26,902] DEBUG: ['0.201250', '0.191625', '0.181250', '0.172531', '0.166250', '0.180250', '0.195000']
[2022-05-12 09:02:26,902] DEBUG: ['0.229109', '0.215378', '0.201188', '0.195915', '0.192000', '0.191380', '0.191219']
[2022-05-12 09:02:26,902] DEBUG: ['0.256875', '0.238828', '0.220625', '0.218922', '0.217500', '0.202109', '0.186875']

Bed Visualizer works great, just don't want it to auto run every time I click the tab. Only when I click the button.

jneilliii, 2022-05-12:
I think I already answered you on Discord, but this is because you don't have the option to save mesh enabled in the plugin's settings.

z00mantwo, 2022-05-12:
That was it. Thanks.

closed 2022-05-12 by jneilliii