jneilliii/OctoPrint-BedLevelVisualizer #543: [Question]: Plugin Support from standard GCode?

JesNoFear, 2022-05-15:
So, this is more of a question, but could turn into a feature request.

I use Cura as my slicer, and at the beginning of every print, it does the bed leveling. Can I add "@BEDLEVELVISUALIZER" into the GCODE of my prints and have it capture the results of the last print?
I am not too sure if it ONLY looks for the "@BEDLEVELVISUALIZER" when running via the Bed Visualizer tab, or any time that command is seen.

It would be nice if any time i looked at the bed visualizer tab, it was the result of the last prints plot points instead of having to separately run this to see if i need to adjust my bed any before my next print.

Thanks and sorry for using this form for a question.
- Jason

jneilliii, 2022-05-15:
> Can I add "@BEDLEVELVISUALIZER" into the GCODE of my prints and have it capture the results of the last print?

Correct.

jneilliii, 2022-05-15:
Just make sure to have it just prior to the G29 command.

closed 2022-05-15 by jneilliii

JesNoFear, 2022-05-15:
Perfect! thank you :-)
thank you for the fast response