jneilliii/OctoPrint-BedLevelVisualizer #545: [BUG]: Display chart  layout does not match display graph

htrabbit, 2022-05-27:
**Describe the bug**
<in the chart Y0 is at the topleft
Image for the Dial is not visible

**Expected behavior**
<i expect to see X0Y0 in the bottom left of the chart to match the image\

**Screenshots**
[image]

**Firmware and Version**
[octoprint-systeminfo-20220527085117.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8787868/octoprint-systeminfo-20220527085117.zip)

jneilliii, 2022-05-27:
There's options in the adjustments section in settings to reverse the rows in the table.

htrabbit, 2022-05-27:
Yes, i see those, they had no effect. i will reinstall and start fresh. i was going to post logs but both logs are empty

  Michael Harvey
[H&N640x480]

     ByteMe Ltd.

***@***.******@***.***>

htrabbit, 2022-05-27:
any ideas about the thumb wheel in the middle of the image instead of the right side?
also here is the logging window, am i doing something wrong?
[image]

jneilliii, 2022-05-28:
The mixed up columns is usually because of uneven probe point quantity x,y. On the support team of the plugin's settings enable debug logging there and restart octoprint.

htrabbit, 2022-05-28:
the octo log shows dubug log set, but i get no bed log.
[octoprint (4).log.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8790174/octoprint.4.log.txt)

jneilliii, 2022-05-28:
If you want to enable debug logging, do it from the plugin's settings as instructed please, not OctoPrint's logging settings.

htrabbit, 2022-05-28:
ok, got the debug log, still can not reverse the Y-position in the adjustment chart so that 0,0 is in the bottom left. i tried with flip y-axis  checked and unchecked. 
[plugin_bedlevelvisualizer_debug.log.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8791886/plugin_bedlevelvisualizer_debug.log.txt)

[image]
[image]
[image]

as for the  thumbwheel being in the middle of the chart, i think it might be a image scale issue because i use octoprint widescreen, it might be pushing the thumbwheel over Maybe.
[image]

jneilliii, 2022-05-28:
Oh sorry, it's on the Data tab. It should adjust both. 

[image]

Based on the log it is as I expected, a rectangular probing data. Seems like you're getting a 13x9 probing grid.

```
['-0.075833', '-0.089080', '-0.110623', '-0.122500', '-0.115444', '-0.100833', '-0.091667', '-0.091056', '-0.091667', '-0.090833', '-0.080988', '-0.064809', '-0.055000']
['-0.095003', '-0.109291', '-0.132563', '-0.145539', '-0.138939', '-0.124842', '-0.116008', '-0.115762', '-0.116875', '-0.116273', '-0.105993', '-0.089027', '-0.078734']
['-0.130312', '-0.146531', '-0.173009', '-0.188021', '-0.182250', '-0.169084', '-0.160854', '-0.161303', '-0.163381', '-0.163229', '-0.152148', '-0.133728', '-0.122542']
['-0.163591', '-0.181718', '-0.211360', '-0.228367', '-0.223313', '-0.210903', '-0.203190', '-0.204452', '-0.207715', '-0.208154', '-0.196324', '-0.176504', '-0.164453']
['-0.176667', '-0.195765', '-0.227012', '-0.245000', '-0.240062', '-0.227642', '-0.220000', '-0.221981', '-0.226407', '-0.227500', '-0.215389', '-0.194944', '-0.182500']
['-0.151065', '-0.170007', '-0.200945', '-0.218549', '-0.212089', '-0.197281', '-0.188464', '-0.192173', '-0.199787', '-0.202906', '-0.191450', '-0.171643', '-0.159544']
['-0.097937', '-0.116263', '-0.146092', '-0.162646', '-0.153242', '-0.133877', '-0.122792', '-0.129566', '-0.142890', '-0.149667', '-0.139551', '-0.121164', '-0.109854']
['-0.043549', '-0.061186', '-0.089784', '-0.105211', '-0.092837', '-0.068888', '-0.055516', '-0.065323', '-0.084309', '-0.094719', '-0.085973', '-0.069058', '-0.058570']
['-0.014167', '-0.031426', '-0.059352', '-0.074167', '-0.060191', '-0.033772', '-0.019167', '-0.030605', '-0.052636', '-0.065000', '-0.056994', '-0.040877', '-0.030833']
```

could you place the `@BEDLEVELVISUALIZER` line just above the `BED_MESH_OUTPUT` line and see if it makes a difference? would like to see what's at the very end of the process in the terminal tab.

htrabbit, 2022-05-28:
Like this?
HOME
BED_MESH_CALIBRATE
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
BED_MESH_OUTPUT		; report the bed leveling mesh points.
SAVE_CONFIG

jneilliii, 2022-05-28:
Yes, like that. If you change to `probe_count: 5,5` it would resolve the wheel issue.

htrabbit, 2022-05-28:
so close , lol. 
No thumbwheel at all now. too many points haha

[image]

htrabbit, 2022-05-28:
ok, if i set
mesh_pps: 0  so there is NO  Mesh Interpolation......Victory
we just lose some "guessed" resolution
[image]

jneilliii, 2022-05-28:
With mesh interpolation enabled what does the command BED_MESH_OUTPUT return on the terminal tab? I suppose it's outputting the points in-between the probe points in it's report, or did you change probe_count to 13,13? I personally am using UBL in Marlin and that is utilizing a 10,10 probe_count.

htrabbit, 2022-05-29:
that was 5,5 probe With no mesh interpolation
in klipper i can enable mesh interpolation with many in between points
ie 
mesh_pps: 2,3  # 2 points between every x and 3 in between every y

what would you like to see and i will run them

htrabbit, 2022-05-29:
with mesh_pps: 2,2
[Bed_lvl_octoprint.txt](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8792430/Bed_lvl_octoprint.txt)
[image]

jneilliii, 2022-05-29:
That makes more sense to me now. So it basically needs to be equal x,y for both probe_count and for mesh_pps.

htrabbit, 2022-05-29:
yes I have confirmed that if I change it to any odd value, the display loses the thumb wheel. the only other thing that would be nice on this, is if we could have positions for the thumb wheels like they do on klipper. as an example I don't have a 169 thumb wheels to adjust. rather I have six three along X the front, and three along X at the rear.

jneilliii, 2022-05-29:
Yeah, the rectangular issue is covered by #381. For what you're asking for it seems like you would benefit more from the Autobim plugin.

htrabbit, 2022-05-29:
i think we have a good solution, keep X andY samples equal.