jneilliii/OctoPrint-BedLevelVisualizer #548: [FR]:

PeterGillham1, 2022-06-18:
**Is your feature request related to a problem? Please describe.**

When printing multiple products (PLA,PLA+,PETG etc.), they all need different bed temperatures.
It would be nice to be able to select the product with pre-defined temperature rather than having to edit the code manually.
*Describe the solution you'd like**

To be able to configure a product with a pre-defined bed temperature and simply select the product rather than changing the code all the time
**Describe alternatives you've considered**

Bed temperature selection dopdown would be OK

**Additional context**

jneilliii, 2022-06-18:
You can technically use the commands to create your own custom buttons and as long as you use the @BEDLEVELVISUALIZER tag it would work. I use this approach for actually doing my probing with UBL and the various slots for different temperatures.

[image]

PeterGillham1, 2022-06-19:
Many thanks - neat solution :-)

closed 2022-06-19 by PeterGillham1