jneilliii/OctoPrint-BedLevelVisualizer #549: [BUG]: Beep before retrieving mesh does not stop.

TLBradbury, 2022-06-18:
**Describe the bug**
Before retrieving the current mesh, the plugin emits a short beep, but sometimes it does not stop. Once it has started, there is no way to silence it.

**Expected behavior**
A short beep before retrieving current mesh.

**Debug Log**

[octoprint-logs.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/8933592/octoprint-logs.zip)

**Screenshots**

**Firmware and Version**
Firmware Name: Marlin 2.0.9.2 (Nov 14 2021 10:06:37)
Machine Type: Ender-3 4.2.7

jneilliii, 2022-06-18:
That's probably just a printer firmware issue. Remove the lines with M300 from your gcode commands and you'll be fine.

closed 2022-06-18 by jneilliii

TLBradbury, 2022-06-18:
Thanks, I'll be sure to give it a try. I was planning on updating my firmware soon anyway (changing hotend).