jneilliii/OctoPrint-BedLevelVisualizer #551: [FR]: Export Mesh Values

ademolay2000, 2022-06-25:
This may already be in here, but the ability to export mesh values directly from the visualizer would be handy (i.e. CSV).  There are times when I'd like to compare the values between runs, and dropping them into Excel would be great.

jneilliii, 2022-06-26:
You might want to try my cmpanion plugin that allows you to switch through previous runs.

https://github.com/jneilliii/OctoPrint-BLVLogger

jneilliii, 2022-06-26:
going to mark this as duplicate of #329

closed 2022-06-26 by jneilliii

ademolay2000, 2022-06-26:
I'm after the raw data for manipulation.  Does BLV store that anywhere accessible to laymen?

jneilliii, 2022-06-26:
Bed Level Visualizer does not, BLV Logger plugin linked above saves the data to sqllite database on the server.