jneilliii/OctoPrint-BedLevelVisualizer #552: [FR]: add 4 buttons which move to a corner and measure the z height for manual tramming

hansvanlin, 2022-06-30:
**To save time with tramming the bed with a paper I use the probe to measure the z height and display it in the printer notification area. Then I use your plugin to measure the mesh and start printing. In this way it is more accurate to tram the bed before starting to update the mesh.**

**Add a button for each corner to probe the corner, below the mesh graph,
[image]
 and measure the z height of that corner**

**Currently I'm using the custom button area to do so but more users could benefit from this method.**

**Additional context**

jneilliii, 2022-06-30:
This is what I made need leveling wizard for. There's also autobim plugin if you have a probe.

hansvanlin, 2022-06-30:
What I was proposing was something of both plugins. Manual measuring 1 corner adjust if necessary again then move to the opposite corner and back to the first one etc.. Both plugins are a wizard which do all corners in sequence which takes time. I could not find a plugin which does have the freedom for the user to choose. I have a probe and want to adjust one corner I choose which needs most adjustment.  And finaly measure all 4 corners before update mesh. Therefore I made the custom buttons which just use G-code commands. But It would nice to have in one place e.g. below the graph.

best regards,
Hans

jneilliii, 2022-06-30:
Oh, so this plugin actually has a hidden feature that allows you to move to the positions if that's what you mean. In the hover menu at the top of the graph there is a move icon. When you click that and then click on a position on the surface it will move the nozzle to that position on the printer.

hansvanlin, 2022-06-30:
Yes something like that and then measure on that point the z height and display it. But this move feature, you have to click twice to move to a new position (on the move feature and then on the bed).  Ideal is when you can set the measuring point exactly above the adjustment screws and have a larger button to click on to move to the predefined corner. This is one click on the button and have the z height displayed after the single measurement. 

I can work how I do it now with the custom buttons but my toughts where that more people would like it.
p.s. you could also make the corners of the mesh data have a button function and update the value in that field, then no extra space is needed.

WireBe, 2022-07-19:
Very intesting. Following

briodan, 2023-06-17:
> What I was proposing was something of both plugins. Manual measuring 1 corner adjust if necessary again then move to the opposite corner and back to the first one etc.. Both plugins are a wizard which do all corners in sequence which takes time. I could not find a plugin which does have the freedom for the user to choose. I have a probe and want to adjust one corner I choose which needs most adjustment. And finaly measure all 4 corners before update mesh. Therefore I made the custom buttons which just use G-code commands. But It would nice to have in one place e.g. below the graph. And have the webcam view when using the buttons.
> 
> best regards, Hans

do you mind sharing the custom buttons code?

hansvanlin, 2023-06-17:
> 

You can add this in the config.yaml located in the .octoprint folder.

controls:
- children:
  - command: M500
    name: Store Settings (M500)
  - commands:
    - G28
    - G29 T
    - M500
    - M117 Bed mesh levels completed
    name: Auto Bed Level
  - command: G30
    name: Measure
  layout: horizontal
  name: Commands
- children:
  - command: G30 P0 X50 Y50
    name: Lower Left
  - command: G30 P0 X220 Y50
    name: Lower Right
  - command: G30 P0 X50 Y220
    name: Upper Left
  - command: G30 P0 X220 Y220
    name: Upper Right
  layout: horizontal
  name: Probe
- children:
  - command: M114
    name: Get Position
    regex: X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)
    template: 'Position: X={0}, Y={1}, Z={2}, E={3}'
  name: Reporting

You also need a plugin to display the z height eg ¨DisplayZ" plugin.

It is also possible to add it in the bed visualizer plugin 
[image]

[image]