jneilliii/OctoPrint-BedLevelVisualizer #554: [BUG]:

stevemcc1980, 2022-07-07:
**Describe the bug**
Send: BEDLEVELVISUALIZER
Recv: echo:Unknown command: "BEDLEVELVISUALIZER"
Recv: ok

**Expected behavior**
Create a mesh and show the results

**Debug Log**
[2022-07-06 17:30:03,280] DEBUG: Canceling mesh collection per user request
[2022-07-06 17:30:03,283] DEBUG: Mesh data collected prior to cancel:
[2022-07-06 17:30:03,283] DEBUG: []
[2022-07-06 17:30:03,284] DEBUG: Mesh data after clearing:
[2022-07-06 17:30:03,284] DEBUG: []
[2022-07-06 17:54:22,013] DEBUG: Canceling mesh collection per user request
[2022-07-06 17:54:22,015] DEBUG: Mesh data collected prior to cancel:
[2022-07-06 17:54:22,016] DEBUG: []
[2022-07-06 17:54:22,016] DEBUG: Mesh data after clearing:
[2022-07-06 17:54:22,017] DEBUG: []
[2022-07-06 18:09:30,046] DEBUG: Canceling mesh collection per user request
[2022-07-06 18:09:30,047] DEBUG: Mesh data collected prior to cancel:
[2022-07-06 18:09:30,048] DEBUG: []
[2022-07-06 18:09:30,048] DEBUG: Mesh data after clearing:
[2022-07-06 18:09:30,048] DEBUG: []
[2022-07-07 16:02:37,522] DEBUG: Canceling mesh collection per user request
[2022-07-07 16:02:37,524] DEBUG: Mesh data collected prior to cancel:
[2022-07-07 16:02:37,524] DEBUG: []
[2022-07-07 16:02:37,525] DEBUG: Mesh data after clearing:
[2022-07-07 16:02:37,525] DEBUG: []

**Screenshots**

What the terminal shows after manually inputting G29 T

Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4
Recv:  0 +0.153 +0.095 +0.060 +0.083 +0.003
Recv:  1 +0.128 +0.085 +0.068 +0.070 +0.005
Recv:  2 +0.055 +0.030 +0.020 +0.078 -0.002
Recv:  3 -0.012 -0.017 -0.032 +0.008 -0.030
Recv:  4 -0.045 -0.097 -0.082 -0.072 -0.080
Recv: 
Recv: Subdivided with CATMULL ROM Leveling Grid:
Recv:         0        1        2        3        4        5        6        7        8        9       10       11       12
Recv:  0 +0.15250 +0.13250 +0.11250 +0.09500 +0.07954 +0.06658 +0.06000 +0.06704 +0.08046 +0.08250 +0.06343 +0.03296 +0.00250
Recv:  1 +0.14593 +0.12751 +0.10910 +0.09333 +0.08042 +0.07016 +0.06454 +0.06901 +0.07813 +0.07759 +0.05940 +0.03155 +0.00370
Recv:  2 +0.13935 +0.12253 +0.10570 +0.09167 +0.08131 +0.07374 +0.06908 +0.07098 +0.07580 +0.07269 +0.05538 +0.03015 +0.00491
Recv:  3 +0.12750 +0.11241 +0.09732 +0.08500 +0.07658 +0.07093 +0.06750 +0.06935 +0.07343 +0.07000 +0.05333 +0.02917 +0.00500
Recv:  4 +0.10667 +0.09350 +0.08032 +0.06972 +0.06223 +0.05732 +0.05593 +0.06282 +0.07323 +0.07389 +0.05710 +0.03054 +0.00398
Recv:  5 +0.08056 +0.06945 +0.05834 +0.04945 +0.04227 +0.03732 +0.03824 +0.05268 +0.07300 +0.08000 +0.06283 +0.03234 +0.00185
Recv:  6 +0.05500 +0.04611 +0.03722 +0.03000 +0.02306 +0.01778 +0.02000 +0.03926 +0.06602 +0.07750 +0.06102 +0.02926 -0.00250
Recv:  7 +0.03083 +0.02535 +0.01986 +0.01482 +0.00793 +0.00149 +0.00278 +0.02150 +0.04795 +0.06028 +0.04649 +0.01857 -0.00935
Recv:  8 +0.00722 +0.00529 +0.00336 +0.00046 -0.00645 -0.01433 -0.01500 +0.00046 +0.02314 +0.03445 +0.02440 +0.00299 -0.01842
Recv:  9 -0.01250 -0.01380 -0.01509 -0.01750 -0.02380 -0.03120 -0.03250 -0.02037 -0.00213 +0.00750 +0.00074 -0.01463 -0.03000
Recv: 10 -0.02592 -0.03151 -0.03709 -0.04176 -0.04633 -0.04998 -0.04935 -0.03977 -0.02590 -0.01842 -0.02302 -0.03401 -0.04500
Recv: 11 -0.03546 -0.04825 -0.06104 -0.06963 -0.07182 -0.06980 -0.06592 -0.05896 -0.05013 -0.04546 -0.04836 -0.05543 -0.06250
Recv: 12 -0.04500 -0.06500 -0.08500 -0.09750 -0.09731 -0.08963 -0.08250 -0.07815 -0.07435 -0.07250 -0.07370 -0.07685 -0.08000
Recv: 
[...]
Recv: //action:prompt_end
Recv: ok

**Firmware and Version**
Marlin 2.0.1

When carrying out a G28 or a G29 T the printer does as requested but the @BEDLEVELVISUALIZER command is not known. Strange thing is the visualizer was working until the 6/7/22 and then suddenly stopped. Nothing else has changed (firmware/plugins etc). Ive removed the plug in and cleared all extras and now cant get any mesh to show.

stevemcc1980, 2022-07-07:
Never mind! I forgot i recently enabled a creality cloud plug-in. Deleted that and now have my meshes back!! Maybe a note for others!

closed 2022-07-07 by stevemcc1980

MadScienzzz, 2022-07-07:
Hi.  Having the exact same issue - what Creality cloud plugin were you using?

jneilliii, 2022-07-07:
There's only one that's called creality cloud.