jneilliii/OctoPrint-BedLevelVisualizer #557: [FR]: Orthogonal rendering

JernejL, 2022-07-14:
A orthogonal rendering would be very welcome, as it would enable better perspective at the bed, allowing to see labels that are obscured due to perspecive vanishing point. 
 
An toggle option for this would be welcome.

JernejL, 2022-07-15:
https://en.wikipedia.org/wiki/Orthographic_projection
 
it is a rendering view with no perspective, it would allow for axis labels to be shown regardless of "depth"

jneilliii, 2022-07-15:
See if this works. Load the tab to show the graph and then open your browser's developer tools and copy/paste the commands below into the console tab and hit enter. 

```
temp1 = $('#bedlevelvisualizergraph')[0].layout;
temp1.scene.camera['projection'] = {'type': 'orthographic'};
Plotly.relayout('bedlevelvisualizergraph', temp1);
```

Changed this

[image]

to this for me

[image]

I see what you mean about being able to see the grid labels, but personally not sure if I like the look. Would definitely consider making it an option.

JernejL, 2022-07-15:
I love it, if that is availible as an option that would be awesome.

jneilliii, 2022-07-16:
Added with the above commit to the rc branch and will be available in next rc release after getting a couple of more issues completed this weekend. You can change the plugin's release channel in OctoPrint's Software Update settings to Release Candidate and update when it comes out, or manually install by copying/pasting the URL below into plugin manager > get more > ...from URL and clicking install.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/heads/rc.zip
```

closed 2022-07-16 by jneilliii