jneilliii/OctoPrint-BedLevelVisualizer #560: [BUG]: Unfortunately it no longer works with 4.4.0-BETA2 Firmware for Original Prusa MINI/MINI+

tafelberg, 2022-07-28:
**Describe the bug**

**Expected behavior**

**Debug Log**

**Screenshots**

**Firmware and Version**
4.4.0-BETA2 Firmware for Original Prusa MINI/MINI+

jneilliii, 2022-07-28:
Maybe they changed the commands? Would need lots more information than it just doesn't work since I don't have a mini. Enable debug logging, serial logging etc and that might help to see what the mini is doing/responding with.

jneilliii, 2022-07-28:
Might be able to see what's happening from the terminal tab. It might show us basically the same info that would need logged to serial.log when enabled.

tafelberg, 2022-07-28:
[serial.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/9213571/serial.log)
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/9213573/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2022-07-28:
weird, I don't see the command being sent to the printer for actually doing the probe operation. Are you using the [Prusa Mini gcode](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/master/wiki/gcode-examples.md#prusa-mini) from the examples?

tafelberg, 2022-07-29:
[image]

Starting with the button (Update Mesh Now) in octoprint. (As always for two years)
Then I see that the print head moves to the first position. The printer heats up. At 59 degrees, the printer beeps once and nothing happens anymore. Without restarting the printer, I can no longer print.

jneilliii, 2022-07-29:
You might want to ask Prusa what has changed in their firmware, I can't control that.

tafelberg, 2022-08-17:
Prusa-Support is now looking for it.

closed 2022-09-09 by github-actions[bot]

tafelberg, 2022-10-02:
Same problem with 4.4.0-RC1
Prusa I also informed
(https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/2322)

jneilliii, 2022-10-02:
curious if you simplify the gcode options down to the following if it works?

```
@BEDLEVELVISUALIZER
G29 T
```

tafelberg, 2022-10-02:
Then nothing happens

tafelberg, 2022-10-08:
This code works. But at the end there is a beep and the plugin crashed. I have always to restart the mini.

M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28 ; home all axes
M420 S0 ; Turning off bed leveling while probing, if firmware is set
; to restore after probing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER ; tell the plugin to watch for reported mesh
G29 T ; run bilinear probing
M500 ; store mesh in EEPROM
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display

[image]

jneilliii, 2022-10-08:
Just take out any additional gcode that is unnecessary, so the M300 commands and M117 commands are beeps and messages on printer screen.