jneilliii/OctoPrint-BedLevelVisualizer #561: Always repeating the gcode

gabrielmaster123, 2022-08-02:
When I started to vidualize the bed the printer just repeats it over and over again. I use Marlin and only changed the bed temp to 70 degrees

jneilliii, 2022-08-02:
Sorry, but I don't understand what you mean.

gabrielmaster123, 2022-08-02:
The printer loops. When it is done scanning the bed it just does it again
over and over again

jneilliii, 2022-08-02:
You might have the wrong commands configured in settings. I suspect you used the UBL commands, but in fact your printer needs the bilinear options.

gabrielmaster123, 2022-08-02:
I have no Idea what either of those mean

jneilliii, 2022-08-03:
Please share the gcode commands you have configured in settings.

gabrielmaster123, 2022-08-03:
GCODE Commands for Mesh Update Process
```G28       ; home all axes
M420 S0   ; Turning off bed leveling while probing, if firmware is set
          ; to restore after G28
M155 S30  ; reduce temperature reporting rate to reduce output pollution
M190 S70  ; (optional) wait for the bed to get up to temperature
G29 P1    ; automatically populate mesh with all reachable points
G29 P3    ; infer the rest of the mesh values
G29 P3    ; infer the rest of the mesh values again
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
M420 S1 V ; enabled leveling and report the new mesh
G29 S0    ; Save UBL mesh points to slot 0 (EEPROM).
G29 F 10.0 ; Set Fade Height for correction at 10.0 mm.
G29 A     ; Activate the UBL System.
M500      ; save the current setup to EEPROM
M155 S3   ; reset temperature reporting
M140 S0   ; cooling down the bed```

gabrielmaster123, 2022-08-03:
ok I figured outr what you meant and took the other code now it works

closed 2022-08-03 by gabrielmaster123