jneilliii/OctoPrint-BedLevelVisualizer #568: [FR]: RF2000 support

gergap, 2022-09-30:
The firmware of my printer (Renkforce RF2000) is based on Repetier.
It supports bed scanning and can output the bed matrix like this using the GCODE command `M3013`.
See also https://www.rf1000.de/wiki/index.php/GCodes

On the Octoprint terminal this looks like this:
```
Send: N10 M3013 P1*98
Recv: ok 10
Recv: M3013: current heat bed z-compensation matrix:
Recv: front left ... front right
Recv: ...        ...         ...
Recv: back left  ...  back right
Recv: ;7;0;20;40;60;80;100;120;140;160;180
Recv: ;0;-0.15;-0.21;-0.22;-0.23;-0.24;-0.28;-0.31;-0.36;-0.40;-0.44
Recv: ;30;-0.15;-0.21;-0.22;-0.23;-0.24;-0.28;-0.31;-0.36;-0.40;-0.44
Recv: ;50;-0.17;-0.22;-0.24;-0.24;-0.25;-0.29;-0.33;-0.38;-0.41;-0.45
Recv: ;70;-0.19;-0.24;-0.25;-0.26;-0.28;-0.32;-0.35;-0.39;-0.42;-0.47
Recv: ;90;-0.21;-0.26;-0.27;-0.28;-0.30;-0.33;-0.36;-0.40;-0.42;-0.48
Recv: ;110;-0.22;-0.27;-0.29;-0.29;-0.30;-0.34;-0.36;-0.40;-0.43;-0.48
Recv: ;130;-0.23;-0.28;-0.29;-0.29;-0.31;-0.34;-0.37;-0.40;-0.44;-0.48
Recv: ;150;-0.25;-0.29;-0.30;-0.30;-0.31;-0.35;-0.37;-0.41;-0.45;-0.50
Recv: ;170;-0.25;-0.29;-0.30;-0.30;-0.31;-0.36;-0.38;-0.42;-0.46;-0.50
Recv: ;190;-0.25;-0.29;-0.30;-0.30;-0.31;-0.35;-0.38;-0.42;-0.45;-0.50
Recv: ;210;-0.24;-0.29;-0.30;-0.30;-0.31;-0.35;-0.38;-0.41;-0.45;-0.50
Recv: ;230;-0.25;-0.29;-0.31;-0.31;-0.33;-0.36;-0.40;-0.43;-0.46;-0.50
Recv: ;245;-0.25;-0.29;-0.31;-0.31;-0.33;-0.36;-0.40;-0.43;-0.46;-0.50
Recv: offset = -193 [steps] (= -0.15 [mm])
Recv: warpage = -453 [steps] (= -0.35 [mm])
Recv: g_uZMatrixMax[X_AXIS] = 10
Recv: g_uZMatrixMax[Y_AXIS] = 13
Recv: g_nActiveHeatBed = 1
[...]
Recv: ok 11
```

`M3013 P1` outputs this in [mm], without the `P1` is in [steps]. I guess [mm] is more useful.

The plugin is not able to understand this. Is there a way to make this working?

When using the commands for "GCODE Commands for Mesh Update Process"
```
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
M3013 P1		; report the bed leveling mesh points.
```
I get this error:
```
TypeError: Failed to execute 'uniformMatrix4fv' on 'WebGLRenderingContext': Overload resolution failed.
Received Data: [object Object]
```
which is most likely because it cannot interpret the firmware output correctly.

jneilliii, 2022-09-30:
It would require some changes I think. Could you enable debug logging in the plugin's settings, restart octoprint and then do an update mesh process. Then share the plugin_bedlevelvisualizer_debug.log file from the logging section of OctoPrint's settings.

gergap, 2022-09-30:
Here is the log file. Doesn't contain much helpful info, though.

[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/9686834/plugin_bedlevelvisualizer_debug.log)

jneilliii, 2022-10-02:
I may have it working with the above commit. If you don't mind installing by copying/pasting the URL below into plugin manager > get more > ...fromURL and clicking Install and reporting back that would be great.

```
https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/archive/refs/heads/rc.zip
```

gergap, 2022-10-02:
Thx a lot. We are almost there ;-)

the current log:
```
[2022-10-02 11:54:24,172] DEBUG: mesh collection started
[2022-10-02 11:54:24,189] DEBUG: mesh collection started
[2022-10-02 11:54:24,249] DEBUG: ['.', '.', '.', '.', '.', '.', '.', '.', '.']
[2022-10-02 11:54:24,256] DEBUG: []
[2022-10-02 11:54:24,260] DEBUG: ['-0.15', '-0.21', '-0.22', '-0.23', '-0.24', '-0.28', '-0.31', '-0.36', '-0.40', '-0.44']
[2022-10-02 11:54:24,264] DEBUG: ['-0.15', '-0.21', '-0.22', '-0.23', '-0.24', '-0.28', '-0.31', '-0.36', '-0.40', '-0.44']
[2022-10-02 11:54:24,268] DEBUG: ['-0.17', '-0.22', '-0.24', '-0.24', '-0.25', '-0.29', '-0.33', '-0.38', '-0.41', '-0.45']
[2022-10-02 11:54:24,273] DEBUG: ['-0.19', '-0.24', '-0.25', '-0.26', '-0.28', '-0.32', '-0.35', '-0.39', '-0.42', '-0.47']
[2022-10-02 11:54:24,277] DEBUG: ['-0.21', '-0.26', '-0.27', '-0.28', '-0.30', '-0.33', '-0.36', '-0.40', '-0.42', '-0.48']
[2022-10-02 11:54:24,288] DEBUG: ['-0.22', '-0.27', '-0.29', '-0.29', '-0.30', '-0.34', '-0.36', '-0.40', '-0.43', '-0.48']
[2022-10-02 11:54:24,292] DEBUG: ['-0.23', '-0.28', '-0.29', '-0.29', '-0.31', '-0.34', '-0.37', '-0.40', '-0.44', '-0.48']
[2022-10-02 11:54:24,296] DEBUG: ['-0.25', '-0.29', '-0.30', '-0.30', '-0.31', '-0.35', '-0.37', '-0.41', '-0.45', '-0.50']
[2022-10-02 11:54:24,300] DEBUG: ['-0.25', '-0.29', '-0.30', '-0.30', '-0.31', '-0.36', '-0.38', '-0.42', '-0.46', '-0.50']
[2022-10-02 11:54:24,304] DEBUG: ['-0.25', '-0.29', '-0.30', '-0.30', '-0.31', '-0.35', '-0.38', '-0.42', '-0.45', '-0.50']
[2022-10-02 11:54:24,308] DEBUG: ['-0.24', '-0.29', '-0.30', '-0.30', '-0.31', '-0.35', '-0.38', '-0.41', '-0.45', '-0.50']
[2022-10-02 11:54:24,312] DEBUG: ['-0.25', '-0.29', '-0.31', '-0.31', '-0.33', '-0.36', '-0.40', '-0.43', '-0.46', '-0.50']
[2022-10-02 11:54:24,316] DEBUG: ['-0.25', '-0.29', '-0.31', '-0.31', '-0.33', '-0.36', '-0.40', '-0.43', '-0.46', '-0.50']
[2022-10-02 11:54:26,123] DEBUG: {'type': 'rectangular', 'x_min': -35.0, 'x_max': 180.0, 'y_min': -18.0, 'y_max': 230.0, 'z_min': 0.0, 'z_max': 200.0}
[2022-10-02 11:54:26,124] DEBUG: stopping mesh collection
[2022-10-02 11:54:26,137] DEBUG: Final mesh:
[2022-10-02 11:54:26,138] DEBUG: ['-0.25', '-0.29', '-0.31', '-0.31', '-0.33', '-0.36', '-0.40', '-0.43', '-0.46', '-0.50']
[2022-10-02 11:54:26,148] DEBUG: ['-0.25', '-0.29', '-0.31', '-0.31', '-0.33', '-0.36', '-0.40', '-0.43', '-0.46', '-0.50']
[2022-10-02 11:54:26,149] DEBUG: ['-0.24', '-0.29', '-0.30', '-0.30', '-0.31', '-0.35', '-0.38', '-0.41', '-0.45', '-0.50']
[2022-10-02 11:54:26,149] DEBUG: ['-0.25', '-0.29', '-0.30', '-0.30', '-0.31', '-0.35', '-0.38', '-0.42', '-0.45', '-0.50']
[2022-10-02 11:54:26,150] DEBUG: ['-0.25', '-0.29', '-0.30', '-0.30', '-0.31', '-0.36', '-0.38', '-0.42', '-0.46', '-0.50']
[2022-10-02 11:54:26,151] DEBUG: ['-0.25', '-0.29', '-0.30', '-0.30', '-0.31', '-0.35', '-0.37', '-0.41', '-0.45', '-0.50']
[2022-10-02 11:54:26,152] DEBUG: ['-0.23', '-0.28', '-0.29', '-0.29', '-0.31', '-0.34', '-0.37', '-0.40', '-0.44', '-0.48']
[2022-10-02 11:54:26,152] DEBUG: ['-0.22', '-0.27', '-0.29', '-0.29', '-0.30', '-0.34', '-0.36', '-0.40', '-0.43', '-0.48']
[2022-10-02 11:54:26,153] DEBUG: ['-0.21', '-0.26', '-0.27', '-0.28', '-0.30', '-0.33', '-0.36', '-0.40', '-0.42', '-0.48']
[2022-10-02 11:54:26,154] DEBUG: ['-0.19', '-0.24', '-0.25', '-0.26', '-0.28', '-0.32', '-0.35', '-0.39', '-0.42', '-0.47']
[2022-10-02 11:54:26,155] DEBUG: ['-0.17', '-0.22', '-0.24', '-0.24', '-0.25', '-0.29', '-0.33', '-0.38', '-0.41', '-0.45']
[2022-10-02 11:54:26,156] DEBUG: ['-0.15', '-0.21', '-0.22', '-0.23', '-0.24', '-0.28', '-0.31', '-0.36', '-0.40', '-0.44']
[2022-10-02 11:54:26,156] DEBUG: ['-0.15', '-0.21', '-0.22', '-0.23', '-0.24', '-0.28', '-0.31', '-0.36', '-0.40', '-0.44']
[2022-10-02 11:54:26,157] DEBUG: ['.', '.', '.', '.', '.', '.', '.', '.', '.']
```
[image]

gergap, 2022-10-02:
Hi can also see data now in the plugin:
[image]

So maybe parsing is OK now, but the negative coordinates seems to be wrong. The printer has a little weird origin. Is it possible that the plugin as troubles with this negative y-coordinate?

jneilliii, 2022-10-02:
Yeah, it's as i expected. The one line `Recv: ...        ...         ...` is getting matched and recording `['.', '.', '.', '.', '.', '.', '.', '.', '.']` as part of the mesh. Will have to figure out a way to remove that or not detect it as mesh data.

gergap, 2022-10-02:
Hmm, not sure where these dots come from.
Here is the Perl code I used to parse the data and plot using GNUplot.
Maybe the Regex works for you too.
[bedlevel.tar.gz](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/9693073/bedlevel.tar.gz)

jneilliii, 2022-10-02:
yeah, that's similar to what I'm already using [here](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/53823de7908e2cacf0f54a4d1564a875d9848b13/octoprint_bedlevelvisualizer/__init__.py#L56), but since I'm supporting all kinds of firmware have to handle it slightly different and can't include the `;`.

gergap, 2022-10-02:
I have no idea why, but it's working now ;-)
[image]

the last log:
```
[2022-10-02 21:29:13,206] DEBUG: mesh collection started
[2022-10-02 21:29:13,228] DEBUG: mesh collection started
[2022-10-02 21:29:13,296] DEBUG: ['.', '.', '.', '.', '.', '.', '.', '.', '.']
[2022-10-02 21:29:13,309] DEBUG: []
[2022-10-02 21:29:13,317] DEBUG: ['0.04', '-0.13', '-0.14', '-0.15', '-0.16', '-0.20', '-0.23', '-0.26', '-0.30', '-0.35']
[2022-10-02 21:29:13,326] DEBUG: ['0.04', '-0.13', '-0.14', '-0.15', '-0.16', '-0.20', '-0.23', '-0.26', '-0.30', '-0.35']
[2022-10-02 21:29:13,334] DEBUG: ['-0.00', '-0.14', '-0.15', '-0.16', '-0.17', '-0.21', '-0.25', '-0.29', '-0.31', '-0.37']
[2022-10-02 21:29:13,342] DEBUG: ['-0.04', '-0.16', '-0.18', '-0.18', '-0.20', '-0.23', '-0.27', '-0.30', '-0.30', '-0.38']
[2022-10-02 21:29:13,350] DEBUG: ['-0.08', '-0.17', '-0.20', '-0.21', '-0.22', '-0.25', '-0.28', '-0.32', '-0.33', '-0.39']
[2022-10-02 21:29:13,362] DEBUG: ['-0.12', '-0.19', '-0.21', '-0.22', '-0.23', '-0.26', '-0.28', '-0.32', '-0.34', '-0.39']
[2022-10-02 21:29:13,372] DEBUG: ['-0.15', '-0.20', '-0.22', '-0.22', '-0.23', '-0.27', '-0.29', '-0.32', '-0.35', '-0.39']
[2022-10-02 21:29:13,382] DEBUG: ['-0.18', '-0.21', '-0.23', '-0.23', '-0.24', '-0.28', '-0.30', '-0.33', '-0.36', '-0.40']
[2022-10-02 21:29:13,386] DEBUG: ['-0.20', '-0.21', '-0.24', '-0.23', '-0.24', '-0.28', '-0.31', '-0.35', '-0.37', '-0.39']
[2022-10-02 21:29:13,391] DEBUG: ['-0.21', '-0.21', '-0.24', '-0.23', '-0.25', '-0.28', '-0.31', '-0.34', '-0.37', '-0.39']
[2022-10-02 21:29:13,395] DEBUG: ['-0.20', '-0.23', '-0.25', '-0.24', '-0.25', '-0.29', '-0.32', '-0.35', '-0.38', '-0.37']
[2022-10-02 21:29:13,399] DEBUG: ['-0.21', '-0.24', '-0.26', '-0.25', '-0.26', '-0.30', '-0.32', '-0.36', '-0.38', '-0.35']
[2022-10-02 21:29:13,403] DEBUG: ['-0.21', '-0.24', '-0.26', '-0.25', '-0.26', '-0.30', '-0.32', '-0.36', '-0.38', '-0.35']
[2022-10-02 21:29:13,416] DEBUG: {'type': 'rectangular', 'x_min': -35.0, 'x_max': 180.0, 'y_min': -18.0, 'y_max': 230.0, 'z_min': 0.0, 'z_max': 200.0}
[2022-10-02 21:29:13,417] DEBUG: stopping mesh collection
[2022-10-02 21:29:13,422] DEBUG: Final mesh:
[2022-10-02 21:29:13,423] DEBUG: ['-0.21', '-0.24', '-0.26', '-0.25', '-0.26', '-0.30', '-0.32', '-0.36', '-0.38', '-0.35']
[2022-10-02 21:29:13,423] DEBUG: ['-0.21', '-0.24', '-0.26', '-0.25', '-0.26', '-0.30', '-0.32', '-0.36', '-0.38', '-0.35']
[2022-10-02 21:29:13,424] DEBUG: ['-0.20', '-0.23', '-0.25', '-0.24', '-0.25', '-0.29', '-0.32', '-0.35', '-0.38', '-0.37']
[2022-10-02 21:29:13,424] DEBUG: ['-0.21', '-0.21', '-0.24', '-0.23', '-0.25', '-0.28', '-0.31', '-0.34', '-0.37', '-0.39']
[2022-10-02 21:29:13,424] DEBUG: ['-0.20', '-0.21', '-0.24', '-0.23', '-0.24', '-0.28', '-0.31', '-0.35', '-0.37', '-0.39']
[2022-10-02 21:29:13,425] DEBUG: ['-0.18', '-0.21', '-0.23', '-0.23', '-0.24', '-0.28', '-0.30', '-0.33', '-0.36', '-0.40']
[2022-10-02 21:29:13,425] DEBUG: ['-0.15', '-0.20', '-0.22', '-0.22', '-0.23', '-0.27', '-0.29', '-0.32', '-0.35', '-0.39']
[2022-10-02 21:29:13,426] DEBUG: ['-0.12', '-0.19', '-0.21', '-0.22', '-0.23', '-0.26', '-0.28', '-0.32', '-0.34', '-0.39']
[2022-10-02 21:29:13,426] DEBUG: ['-0.08', '-0.17', '-0.20', '-0.21', '-0.22', '-0.25', '-0.28', '-0.32', '-0.33', '-0.39']
[2022-10-02 21:29:13,427] DEBUG: ['-0.04', '-0.16', '-0.18', '-0.18', '-0.20', '-0.23', '-0.27', '-0.30', '-0.30', '-0.38']
[2022-10-02 21:29:13,427] DEBUG: ['-0.00', '-0.14', '-0.15', '-0.16', '-0.17', '-0.21', '-0.25', '-0.29', '-0.31', '-0.37']
[2022-10-02 21:29:13,427] DEBUG: ['0.04', '-0.13', '-0.14', '-0.15', '-0.16', '-0.20', '-0.23', '-0.26', '-0.30', '-0.35']
[2022-10-02 21:29:13,428] DEBUG: ['0.04', '-0.13', '-0.14', '-0.15', '-0.16', '-0.20', '-0.23', '-0.26', '-0.30', '-0.35']
[2022-10-02 21:29:13,428] DEBUG: ['.', '.', '.', '.', '.', '.', '.', '.', '.']
```

gergap, 2022-10-02:
Thanks for your help. If I should test more and send you a log, please let me know.
I'm happy now :+1: