jneilliii/OctoPrint-BedLevelVisualizer #578: [FR]: Bed Visualizer on OctoDash Touchscreen Display as option.

DirtyKityLitter, 2022-12-07:
**Is your feature request related to a problem? Please describe.**

I am frustrated there is not an option that is easily available to view BedLevelVisualizer as a plugin within OctoDash for Touchscreen Displays

**Describe the solution you'd like**

It would be amazing to have a plugin for OctoDash Touchscreens to see BedLevelVisualizer on a touchscreen display instead of using a laptop / pc connected to OctoPrint. I am using Raspberri Pi 7" Touchscreen FYI.

**Describe alternatives you've considered**

I did get your response using the [!WEB[ link.

**Additional context**

jneilliii, 2022-12-07:
Uh, is my `[!WEB]` option with the special URL not good enough? Use that on a custom action in OctoDash settings. 

`[!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer`

DirtyKityLitter, 2022-12-07:
I added the [!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer

As a new custom setting in octodash with a bed as the logo.

When I press it on my touchscreen lcd it does nothing.

Could be I do not fully understand how the [!WEB] command functions either.

I put the whole line of [!WEB]http://127.0.0.1:5000/plugin/bedlevelvisualizer/bedlevelvisualizer

In the line

Should I have just put [!WEB] in there?

Thank you for your help I’m sure these questions / issues get annoying.

I’m willing to make a video of how I got mine working when finished and send it to you for other people with these issues.

jneilliii, 2022-12-07:
That's odd, it should open a new panel with a simplified version of the graph (not in 3d) and a button. 

[image]

[image]

closed 2022-12-29 by github-actions[bot]