jneilliii/OctoPrint-BedLevelVisualizer #579: [BUG]: Hover is not working for Adjustment Angle

Sleeepy2, 2022-12-21:
**Describe the bug**
When hovering over a point on the Adjustment angle (degree) it should the rotation amount. Now it doesn't show anything.

**Expected behavior**
When hovering over a point on the Adjustment angle (degree) it should the rotation amount.

**Screenshots**
[image]

**Firmware and Version**
OctoPrint 1.87.6
Python 3.9.2
OctoPi 1.0.0 RC3

jneilliii, 2022-12-22:
this appears to have been a plugin conflict with Tasmota and TPLinkSmartplug plugins bundled ploty libraries. I have released new rc versions on those plugins to resolve the conflict, and hover should work again, but only on the graph part, not the table below. points on table below will be highlighted while hovering the graph. still contemplating removal of the table completely in favor of the hover popup on the mesh.

Ltek, 2023-02-03:
I dont think this is solved. I do not have any Adjustment Angle showing on the visual. Only if I open Settings (gear icon) and look at the raw data table ( which is also confusing since bottom left is no 0,0 coordinates; which is the math standard.

[image]

jneilliii, 2023-02-03:
@Ltek are you running the rc version? In OctoPrint's Software Update settings change the release channel to `Release Candidate` for the plugin. 

As for your other issue (seems you edited it out so probably found the setting) descending y option resolves that.

Ltek, 2023-02-03:
I didnt realize we need to turn on all that stuff in settings 👍 
BUT... I'm not sure how accurate that table is, my bed is darn level but table shows major adjustments needed?
[image]

jneilliii, 2023-02-03:
did you have to use relative z offset setting to get the graph to show in the view? I'm assuming you may have, because I think the adjustment angles are based on getting to 0 (I didn't program that bit of the plugin). Looking at the numbers it's close to a full turn on all of them.

Ltek, 2023-02-03:
Nope. Dont even know what 'relative z offset' means or does. my settings screenshot...
[image]

jneilliii, 2023-02-03:
Not sure where those high calculations are coming from then, do they make sense relative to the mesh data itself? I've always thought the adjustment calculations didn't make sense unless probe points landed directly on top of you screws.