jneilliii/OctoPrint-BedLevelVisualizer #589: [FR]:

LazyEpic, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
Klipper presents bed mesh in reverse order meaning visualization is flipped on both X and Y (not Z) 

**Describe the solution you'd like**
Would be great with an option to select to flip the X and/or Y axis

**Describe alternatives you've considered**
None really, it's fairly straight forward. 

**Additional context**
This would make the plugin adaptable to most ways of presenting a mesh

jneilliii, 2023-01-29:
You must have missed the flip x and flip y options in the plugin's settings?

LazyEpic, 2023-01-29:
Where? I can flip them in data but that doesn't do anything to the visualization even when reloading the mesh or am I missing something obvious? ^^

closed 2023-01-29 by LazyEpic