jneilliii/OctoPrint-BedLevelVisualizer #591: [BUG]: Homing Z before bed is heated up in Bilinear Leveling

iluvatyr, 2023-02-09:
**Describe the bug**
When using the plugin to create a mesh of the bed, it starts warming up, then homes all axis and then waits until bed temperature is reached, not taking into account the differences, due to the warming up of components and the bed specifically.
I am not sure if it is a bug or if it is intended to be this way.

**Current bilinear leveling Gcode **
`M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M420 S0  ; Turning off bed leveling while probing, if firmware is set
         ; to restore after G28
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up`

**Expected behavior**
In my opinion, it should heat up the bed, and only afterwards home all axis, which can be easily obtained by changing the first line of GCODE from a M140 S60 to a M190 S60.

**My suggestion Gcode**
`M190 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M420 S0  ; Turning off bed leveling while probing, if firmware is set
         ; to restore after G28
M117 Heating the bed ; send message to printer display`

jneilliii, 2023-02-09:
This is exactly why it's an open ended GCODE input area, you can put the commands in any way you feel appropriate.

jneilliii, 2023-02-09:
adjusted in the above commit and will go live with next release.