jneilliii/OctoPrint-BedLevelVisualizer #596: [FR]: Add a manual correction for bed tilt difference between probing and printing

JoeScotty, 2023-02-22:
**Is your feature request related to a problem? Please describe.**
At my printer the print bed is guided using vertical linear rails at the rear side of the bed.
The probe is a BLTouch, which is used for auto levelling.
During printing at the front side of the bed it is tilted down by about 0.1mm compared the the results of bed levelling.
The reason for this effect is the difference between extrusion force on the bed compared to the BLTouch probing force.
Found this by printing one line along the bed limits and measuring the line thicknesses at several positions.

**Describe the solution you'd like**
If I once know the tilting value (found the 0,1mm) this value could be entered manually.
The algorithm adds this value to the front correction positions and decending linear to the rear positions.
The tilting value will probably depend on extrusion width, layer height, hotend temperature, filament type.
Therefore it would be good to have the chance to perform bed levelling once and lateron use several tilting values to be calculated on the bed levelling data and print with them.

**Describe alternatives you've considered**
Save the bed levelling data to a file, read them by a selfmade software which adds the tilting value factor and save the new correction data to a gcode file, which will be sent to the printer.
OR:
use a weight and hang it on the bed during bed levelling ;-)

**Additional context**
Being not sure how often this feature can be used (depend on printer mechanics) and if its worth realizing it.
But let me say I am always happy working with your plugin which is soo helpful!
If I think to the amount of work in former times - getting all values manually, transferring to excel and create the chart  - big progress now.
Thanks in advance!

jneilliii, 2023-02-23:
This is something that I unfortunately will not do. Modifying the mesh or real-time adjustment of gcode is outside the scope of the project. You may want to consider looking at the [UBL Mesh Edit](https://plugins.octoprint.org/plugins/ublmeshedit/)  plugin, which reminds me there only way you'll achieve what you propose is with UBL or manual mesh leveling.

JoeScotty, 2023-02-23:
Thanks for your quick reply!
your tip regarding UBL Mesh Edit is helpful.
And if you have any account where I can buy a beer, let me know!
As said, I find your plugin really helpful!

jneilliii, 2023-02-23:
Thanks for the kind words and understanding on why I wouldn't make this change. If you are inclined, you can send one-time GitHub sponsor [here](https://github.com/sponsors/jneilliii).

closed 2023-02-23 by jneilliii