jneilliii/OctoPrint-BedLevelVisualizer #598: [BUG]:

m4sis, 2023-03-01:
hello,

The image speaks for itself, the table must have a fixed position I think I can't see the values
[image]

jneilliii, 2023-03-01:
What's your screen resolution? My chart doesn't cut off quite that much, but is still cutting off. I suppose I never noticed since I'm using UI Customizer and have the tabs resized. 

Without UI Customizer
[image]

With UI Customizer (3 left column, 9 middle, 0 right)
[image]

With UI Customizer and Consolidated Tabs (tabs panelized into single page taking up half the "middle" column)
[image]

jneilliii, 2023-03-01:
It may also be something that I've already fixed in the rc version of the plugin. If you change the release channel in OctoPrint's software update settings to release candidate and update when prompted you could double-check to see if it's any better.

m4sis, 2023-03-01:
I updated the version but it didn't change anything.