jneilliii/OctoPrint-BedLevelVisualizer #600: [FR]: Klipper Support

EddyMI3d, 2023-03-05:
**Is your feature request related to a problem? Please describe.**
No, no problem, just an extension of possibilities.

**Describe the solution you'd like**
Klipper's output of a bed mesh is quite versatile.

There is the usual output. In difference to Marlin, it adds an interpolated output, depending on the parameters.
This mesh list appends to the usual output. Sadly the whole output does not contain the area of the mesh.

```
Send: BED_MESH_OUTPUT PGP=0 
Recv: // Mesh Leveling Probed Z positions:
Recv: // 0.154500 0.185500 0.167000 0.154500 0.132500 0.122500 0.155250 0.174000 0.161750 0.155250 0.177000
Recv: // 0.043250 0.030500 0.008500 -0.003500 -0.024000 -0.034000 -0.025000 0.026250 0.008750 -0.056750 0.025750
Recv: // -0.039500 -0.019500 -0.028250 -0.053750 -0.066500 -0.067250 -0.069750 -0.034250 -0.034250 -0.054500 -0.064500
Recv: // -0.047750 -0.020250 -0.026750 -0.032250 -0.045750 -0.031750 -0.026500 -0.002000 0.024500 0.021000 -0.057750
Recv: // -0.047250 -0.027750 -0.027750 -0.035750 -0.045750 -0.049000 -0.045500 -0.030500 -0.007000 -0.018000 -0.101250
Recv: // -0.026500 0.005250 -0.006250 -0.035250 -0.059500 -0.067500 -0.069000 -0.058000 -0.053750 -0.066000 -0.135250
Recv: // -0.033000 -0.006750 -0.014000 -0.045500 -0.072250 -0.072750 -0.074000 -0.059500 -0.044000 -0.043500 -0.089750
Recv: // -0.058000 -0.043000 -0.063750 -0.093750 -0.119000 -0.110250 -0.089750 -0.065750 -0.036250 -0.007000 -0.032500
Recv: // -0.066500 -0.085250 -0.095750 -0.121750 -0.172250 -0.161000 -0.118750 -0.116500 -0.113750 -0.073750 -0.056000
Recv: // -0.016750 -0.044000 -0.061000 -0.085500 -0.146000 -0.130500 -0.078250 -0.071500 -0.071250 -0.003750 0.042750
Recv: // 0.034750 0.015750 -0.013000 -0.026000 -0.056250 -0.039500 -0.010000 0.012750 0.037500 0.106500 0.106500
Recv: Mesh X,Y: 41,41
Recv: Search Height: 10
Recv: Mesh Offsets: X=0.0000, Y=0.0000
Recv: Mesh Average: -0.03
Recv: Mesh Range: min=-0.1722 max=0.1855
Recv: Interpolation Algorithm: bicubic
Recv: Measured points:
Recv:   0.034750  0.031695  0.025969  0.019884  0.015750  0.010306  0.001225  -0.007781  -0.013000  -0.015800  -0.019463  -0.023144  -0.026000  -0.031816  -0.041869  -0.051549  -0.056250  -0.054446  -0.049369  -0.043545  -0.039500  -0.034080  -0.024900  -0.015645  -0.010000  -0.005421  0.001494  0.008349  0.012750  0.017074  0.023969  0.031441  0.037500  0.050271  0.072619  0.094657  0.106500  0.108441  0.108225  0.107147  0.106500
Recv:   0.026204  0.022863  0.016611  0.010025  0.005680  0.000562  -0.007835  -0.016163  -0.021074  -0.024117  -0.028351  -0.032696  -0.036073  -0.042886  -0.054681  -0.066080  -0.071710  -0.070127  -0.065155  -0.059317  -0.055139  -0.048985  -0.038437  -0.027847  -0.021564  -0.017313  -0.011240  -0.005331  -0.001572  0.001920  0.007423  0.013537  0.018867  0.031437  0.053666  0.075743  0.087862  0.091274  0.093791  0.095445  0.096270
Recv:   0.010244  0.006376  -0.000849  -0.008362  -0.013094  -0.017606  -0.024728  -0.031792  -0.036131  -0.039626  -0.044923  -0.050502  -0.054844  -0.063507  -0.078530  -0.093116  -0.100469  -0.099297  -0.094522  -0.088662  -0.084238  -0.076723  -0.063641  -0.050575  -0.043112  -0.039472  -0.034964  -0.030815  -0.028250  -0.026302  -0.023383  -0.019789  -0.015812  -0.003623  0.018368  0.040503  0.053125  0.059258  0.066833  0.073546  0.077094
Recv:   -0.006338  -0.010716  -0.018880  -0.027285  -0.032384  -0.036273  -0.042093  -0.047869  -0.051623  -0.055571  -0.061939  -0.068760  -0.074068  -0.084558  -0.102769  -0.120501  -0.129555  -0.128800  -0.124237  -0.118375  -0.113717  -0.104866  -0.089288  -0.073787  -0.065167  -0.062150  -0.059227  -0.056855  -0.055491  -0.055079  -0.054729  -0.053642  -0.051023  -0.039271  -0.017623  0.004459  0.017513  0.026295  0.038811  0.050471  0.056684
Recv:   -0.016750  -0.021359  -0.029950  -0.038753  -0.044000  -0.047512  -0.052569  -0.057591  -0.061000  -0.065198  -0.072163  -0.079670  -0.085500  -0.096922  -0.116750  -0.136078  -0.146000  -0.145479  -0.141069  -0.135249  -0.130500  -0.120984  -0.104156  -0.087438  -0.078250  -0.075602  -0.073575  -0.072198  -0.071500  -0.071899  -0.072894  -0.073129  -0.071250  -0.059866  -0.038656  -0.016868  -0.003750  0.006286  0.021188  0.035245  0.042750
Recv:   -0.026984  -0.031444  -0.039754  -0.048257  -0.053295  -0.056502  -0.061045  -0.065572  -0.068731  -0.072988  -0.080145  -0.087858  -0.093780  -0.105094  -0.124693  -0.143795  -0.153617  -0.153201  -0.149024  -0.143458  -0.138873  -0.129523  -0.112965  -0.096525  -0.087529  -0.085153  -0.083585  -0.082661  -0.082220  -0.082703  -0.083782  -0.084143  -0.082473  -0.071904  -0.052185  -0.031924  -0.019727  -0.010405  0.003433  0.016487  0.023455
Recv:   -0.043125  -0.047210  -0.054820  -0.062595  -0.067175  -0.069923  -0.073740  -0.077575  -0.080375  -0.084664  -0.092002  -0.099885  -0.105812  -0.116609  -0.135227  -0.153360  -0.162700  -0.162407  -0.158628  -0.153533  -0.149294  -0.140487  -0.124866  -0.109369  -0.100931  -0.098951  -0.097950  -0.097550  -0.097375  -0.097743  -0.098526  -0.098679  -0.097156  -0.088026  -0.071019  -0.053586  -0.043175  -0.035642  -0.024641  -0.014315  -0.008806
Recv:   -0.058516  -0.062131  -0.068867  -0.075742  -0.079780  -0.082103  -0.085288  -0.088528  -0.091019  -0.095339  -0.102835  -0.110856  -0.116752  -0.126906  -0.144318  -0.161258  -0.169995  -0.169804  -0.166421  -0.161812  -0.157943  -0.149781  -0.135286  -0.120913  -0.113118  -0.111471  -0.110893  -0.110858  -0.110842  -0.110968  -0.111248  -0.110992  -0.109512  -0.101811  -0.087539  -0.072978  -0.064411  -0.058867  -0.051067  -0.043830  -0.039977
Recv:   -0.066500  -0.069683  -0.075612  -0.081673  -0.085250  -0.087371  -0.090319  -0.093357  -0.095750  -0.100122  -0.107750  -0.115878  -0.121750  -0.131424  -0.147931  -0.163973  -0.172250  -0.172098  -0.168944  -0.164630  -0.161000  -0.153311  -0.139650  -0.126102  -0.118750  -0.117194  -0.116638  -0.116575  -0.116500  -0.116330  -0.116069  -0.115335  -0.113750  -0.106839  -0.094125  -0.081223  -0.073750  -0.069519  -0.063875  -0.058731  -0.056000
Recv:   -0.066646  -0.068756  -0.072695  -0.076788  -0.079356  -0.081605  -0.085125  -0.088820  -0.091594  -0.096186  -0.104119  -0.112471  -0.118322  -0.127187  -0.142153  -0.156649  -0.164108  -0.163869  -0.160822  -0.156723  -0.153328  -0.146367  -0.134026  -0.121744  -0.114962  -0.112901  -0.111184  -0.109851  -0.108943  -0.107791  -0.105916  -0.103647  -0.101310  -0.094603  -0.082551  -0.070475  -0.063695  -0.060963  -0.057971  -0.055450  -0.054128
Recv:   -0.064119  -0.064311  -0.064696  -0.065278  -0.066063  -0.068620  -0.073311  -0.078328  -0.081863  -0.086854  -0.095318  -0.104053  -0.109863  -0.117323  -0.129609  -0.141419  -0.147450  -0.147032  -0.144119  -0.140340  -0.137325  -0.131622  -0.121569  -0.111476  -0.105656  -0.102602  -0.098653  -0.094893  -0.092406  -0.089464  -0.084673  -0.079613  -0.075869  -0.069301  -0.058025  -0.047008  -0.041212  -0.040951  -0.042384  -0.044249  -0.045288
Recv:   -0.060657  -0.058926  -0.055747  -0.052808  -0.051800  -0.054650  -0.060484  -0.066796  -0.071075  -0.076463  -0.085454  -0.094574  -0.100347  -0.106427  -0.116080  -0.125250  -0.129880  -0.129297  -0.126535  -0.123093  -0.120472  -0.116054  -0.108335  -0.100478  -0.095648  -0.091613  -0.085449  -0.079279  -0.075229  -0.070535  -0.062893  -0.055109  -0.049993  -0.043574  -0.033081  -0.023123  -0.018311  -0.020503  -0.026327  -0.032548  -0.035928
Recv:   -0.058000  -0.055180  -0.049981  -0.045041  -0.043000  -0.045928  -0.052250  -0.059134  -0.063750  -0.069347  -0.078637  -0.087984  -0.093750  -0.099095  -0.107344  -0.115109  -0.119000  -0.118371  -0.115769  -0.112595  -0.110250  -0.106641  -0.100381  -0.093930  -0.089750  -0.085250  -0.077975  -0.070588  -0.065750  -0.060187  -0.051131  -0.042010  -0.036250  -0.030062  -0.020250  -0.011125  -0.007000  -0.010640  -0.019019  -0.027763  -0.032500
Recv:   -0.053447  -0.050146  -0.044048  -0.038183  -0.035580  -0.038054  -0.043692  -0.049933  -0.054216  -0.059828  -0.069261  -0.078772  -0.084615  -0.089881  -0.097966  -0.105582  -0.109441  -0.109042  -0.106898  -0.104242  -0.102309  -0.099509  -0.094663  -0.089619  -0.086225  -0.082012  -0.075030  -0.067899  -0.063243  -0.057938  -0.049315  -0.040680  -0.035335  -0.030061  -0.021866  -0.014396  -0.011299  -0.015896  -0.025764  -0.035933  -0.041431
[... cut ...]
Recv: // 118 | (310.0, 350.0)  | (302.0, 390.0)
Recv: // 119 | (344.0, 350.0)  | (336.0, 390.0)
Recv: // 120 | (378.0, 350.0)  | (370.0, 390.0)
Recv: ok
```

**Describe alternatives you've considered**
As a workaround maybe a manual upload feature can give the ability to use the interpolated mesh.

**Additional context**
As you see, the mesh is huge. It's a T-Rex 3 wit an 400mm x 400mm are and 11 x 11 measuring points.
So I do the bed mesh once in a while or when I fumbled too much at the bed :-)

jneilliii, 2023-03-05:
Klipper is supported based on previous feature requests. If it's not working for you using the example commands identified in the wiki then you need to enable debug logging in the plugin settings, restart OctoPrint, try again and then provide logs.

EddyMI3d, 2023-03-05:
Can't for what reason ever upload the file here.
Can I send it to you as PM in the OctoPrint forum?

jneilliii, 2023-03-05:
once enabled and octoprint restarted you should get additional info in plugin_bedlevelvisualizer_debug.log when you attempt an update mesh process.

EddyMI3d, 2023-03-05:
Never mind, it seems my mesh with 11 x 11 is a bit too large.
Did one with 5 x 5 and it went well.

jneilliii, 2023-03-05:
That's odd, the plugin shouldn't really care how many probe points there are. I know with my CR-10 I have 10x10 and it works fine.

EddyMI3d, 2023-03-05:
It's because the last sequence is not properly sent by Klipper. 
It just sent 30 and a half line instead of 42.

jneilliii, 2023-03-05:
No, don't set debug there. Check the box in the plugin's settings on the support tab and restart OctoPrint.

EddyMI3d, 2023-03-06:
Ah, I see.

closed 2023-03-21 by EddyMI3d