jneilliii/OctoPrint-BedLevelVisualizer #601: [BUG]: Prusa mini example does not work

Yourigh, 2023-03-18:
**Describe the bug**
When I copy paste the code from wiki/gcode-examples.md and run bed levelling, I get incomplete table.

**Expected behavior**
Complete bed levelling table.

**Debug Log**
Serial output log is below. I think that the output of the G28 command is interfering with the table data. 
```
Send: G28
[...]
Recv: X:176.40 Y:24.10 Z:2.00 E:0.00 Count X:17640 Y:2410 Z:800
Recv: ok
Send: G29
[...]
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3
Recv:  0 +0.211 +0.156 +0.048 -0.096
Recv:  1 +0.196 +0.317 +0.242 +0.143
Recv:  20.389 +0.421
```

When I run it manually I get complete table. When I adjust the collection gcode like this, I get complete result:
```
M104 S170		; set extruder temp for bed leveling
M140 S60		; set bed temp
M109 R120		; wait for bed leveling temp
G28   ; home all without mesh bed level
M109 R170		; wait for bed leveling temp
M190 S60		; wait for bed temp
;G28			; home all without mesh bed level
@BEDLEVELVISUALIZER	; instruct plugin to start recording responses from printer.
G29			; mesh bed leveling
M104 S0			; cool down head
M140 S0			; cooling down the bed
```
This practically added some "wait time" between G28 and G29, while the hotend gets from 120 to 170. 

**Screenshots**

**Firmware and Version**
Prusa mini 4.4.1

jneilliii, 2023-03-18:
that's odd, never had anyone report a timing issue with G28/G29. other options might be using dwell (G4) command rather than the temp trick you figured out.

Yourigh, 2023-03-18:
I edited the issue. 
In fact, when I run it manually it also did not work every time. 

I found a command that can report the last mesh bed levelling, and so fat that works well.  The config is in original post.

closed 2023-04-04 by Yourigh