jneilliii/OctoPrint-BedLevelVisualizer #605: Unable to Install

jneilliii, 2023-04-03:
I don't think so. How can I activate OctoPrint venv? I simply flashed Octopi to a raspi 4 with everything pre-installed

Thanks for the hint, on Octoprint web I was able to install a plugin. However the BedLevelVisualizer plugin still fails.

_Originally posted by @Youras in https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/345#issuecomment-1494286689_

Youras, 2023-04-04:
Now, after a second try it just worked.

perhaps the previous installation of another plugin resolved some dependencies that were missing before?

Thank you for your feedback

jneilliii, 2023-04-04:
The plugin doesn't have any dependencies outside of what's already bundled with OctoPrint, so unsure of what could have caused the error. Glad you're up and running now.

closed 2023-04-04 by jneilliii