jneilliii/OctoPrint-BedLevelVisualizer #606: [BUG]: Issue running bed visualizer. "IndexError: tuple index out of range"

llacoste, 2023-04-08:
**Describe the bug**
I am attempting to "Update Mesh Now" on a new to me PrintrBot Simple Metal Extended. I am using the sample GCode for Marlin firmware: 

```
M140 S60 ; starting by heating the bed for nominal mesh accuracy
M117 Homing all axes ; send message to printer display
G28      ; home all axes
M420 S0  ; Turning off bed leveling while probing, if firmware is set
         ; to restore after G28
M117 Heating the bed ; send message to printer display
M190 S60 ; waiting until the bed is fully warmed up
M300 S1000 P500 ; chirp to indicate bed mesh levels is initializing
M117 Creating the bed mesh levels ; send message to printer display
M155 S30 ; reduce temperature reporting rate to reduce output pollution
@BEDLEVELVISUALIZER	; tell the plugin to watch for reported mesh
G29 T	   ; run bilinear probing
M155 S3  ; reset temperature reporting
M140 S0 ; cooling down the bed
M500 ; store mesh in EEPROM
M300 S440 P200 ; make calibration completed tones
M300 S660 P250
M300 S880 P300
M117 Bed mesh levels completed ; send message to printer display
```

When I start the process of updating the mesh, the printer heats up, homes the axes, and seems to start the leveling process. The top right and top left corners are successfully traversed but once we get to the lower left corner the process stops. No error is reported to the Octoprint UI but if I pull the Octoprint logs I see the following error:

```
2023-04-07 21:06:31,170 - octoprint.util.comm - ERROR - Error while processing hook bedlevelvisualizer:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/comm.py", line 4040, in _readline
    ret = hook(self, ret)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint_bedlevelvisualizer/__init__.py", line 390, in process_gcode
    z[i][j] = a[2][k]
IndexError: tuple index out of range
```

Any help would be greatly appreciated! 

There are many of these errors.

**Expected behavior**
The bed level mesh is updated successfully.

**Debug Log**
[plugin_bedlevelvisualizer_debug (1).log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/11182336/plugin_bedlevelvisualizer_debug.1.log)

**Screenshots**

**Firmware and Version**
```
Send: M115
Recv: FIRMWARE_NAME:PB Marlin Rev F v5 FIRMWARE_URL:https://github.com/Printrbot/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Printrbot EXTRUDER_COUNT:3 UUID:00000000-0000-0000-0000-000000000000
```

jneilliii, 2023-04-08:
Could you do me a favor and enable serial logging in OctoPrint's settings, and manually run the leveling commands in terminal and then turn serial logging back off and then share the serial.log from the logging section of OctoPrint's settings?

jneilliii, 2023-04-08:
After looking at the firmware that you have, I think the issue is G29 only does a 3 point probing process. 

> G29 - Detailed Z-Probe, probes the bed at 3 points. You must de at the home position for this to work correctly.

This doesn't return the necessary amount of data for the plugin to be able to graph. It requires at least 4 points in order to be able to visualize the bed.

llacoste, 2023-04-08:
Thanks, @jneilliii! I'll work on upgrading the firmware!

jneilliii, 2023-04-08:
In theory, it would be commenting the line [here](https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2/config/examples/Printrbot/Simple%20Metal%20RevD/Configuration.h#L1876) out with a `#` and removing the `#` from [this](https://github.com/MarlinFirmware/Configurations/blob/release-2.1.2/config/examples/Printrbot/Simple%20Metal%20RevD/Configuration.h#L1878) line to enable bilinear leveling.

closed 2023-05-01 by github-actions[bot]