jneilliii/OctoPrint-BedLevelVisualizer #609: [FR]: Is there a command line to run the overall BedLevelVisualiser using http for Farm Owner ?

patisa06, 2023-04-22:
Dear dev,
we are looking for a way to upgrade the bed info for all the farms like:
curl -s -H "Content-Type: application/json" -H "X-Api-Key:$ApiKey" -X POST -d '{ "command":"???????" }'  @ADDRESS
but didn't fine the command ???????
for example connect, disconnect..
Is there a bedleveling that will run the process ?
Best regards
Patrick

jneilliii, 2023-04-22:
You can use the default OctoPrint API to send the same commands configured in the plugin settings.

patisa06, 2023-04-22:
Thanks a lot for the message. What is the command to send to run "Update Mesh Now" like the button in the application ?

jneilliii, 2023-04-22:
Literally what I just said. It does a send Command as if from the control tab of all the lines configured in the plugin's setting. There is no direct API to do the same thing. There is a custom action handler though, where if your printers support it you can use the action `BEDLEVELVISUALIZER_LEVELBED`.

https://marlinfw.org/docs/gcode/M118.html

So you could use the OctoPrint API to send the single M118 command.

jneilliii, 2023-04-22:
For clarity, the full command would be `M118 A1 action:BEDLEVELVISUALIZER_LEVELBED`. Test that from the terminal tab and see if anything is returned. If it is not, then your printer doesn't support HOST_ACTION_COMMANDS.

patisa06, 2023-04-22:
thanks a lot. now we realized that our Prusa Mk3 doesn't support M118 commands. Sorry. i have to do the bedlevel manually so.
Thanks again

jneilliii, 2023-04-22:
I've added an API POST command to start the leveling process remotely. It's available in 1.1.2rc5 version, you can change the release channel in OctoPrint's Software Update settings to `Release Candidate` for the plugin and update when prompted. 

URL = `http://octopi.local/api/plugin/bedlevelvisualizer`
PAYLOAD = `{"command": "startProcessing"}`

patisa06, 2023-04-22:
great !! thanks a lot. you save me !

closed 2023-04-22 by jneilliii