jneilliii/OctoPrint-BedLevelVisualizer #614: [BUG]: Creality CR-10s w/Marlin 2.1.2 and BedLevelVisualizer v/1.1.2rc5

Icegames-dk, 2023-05-10:
**Describe the bug**
BLV stops in the middel of the bed, with no errors.

**Debug Log**
[plugin_bedlevelvisualizer_debug.log](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/11440842/plugin_bedlevelvisualizer_debug.log)

**Screenshots**
[image]
[image]

**Firmware and Version**
Creality CR-10s w/Marlin 2.1.2 and BedLevelVisualizer v/1.1.2rc5

jneilliii, 2023-05-10:
Not sure I understand what the issue is. Does it not generate a graph? The debug log seems to indicate that it does receive a proper mesh. One thing I notice is that you are using UBL commands, is that something you enabled in your own custom compile of Marlin or are you using a Creality provided firmware or other pre-built fork? It's fairly uncommon for manufacturers to use UBL and they typically enable bilinear instead.

Icegames-dk, 2023-05-11:
Hi
It's the standard firmware from Marlin I used.
https://marlinfw.org/meta/download/

And this is the configuration files that they provide for Creality CR10s
[Creality CR10s.zip](https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/files/11449675/Creality.CR10s.zip)

Please bear with me I am new to Octoprint :-)

closed 2023-06-02 by github-actions[bot]