jneilliii/OctoPrint-BedLevelVisualizer #616: [BUG]: for klipper: new GCODE Commands for Mesh Update Process

leo-na-rd, 2023-05-26:
**Describe the bug**
cannot start mesh probing

**Expected behavior**
probe then display mesh

**Firmware and Version**
Version 1.1.1

***Solution***
G28
BED_MESH_CALIBRATE 
BED_MESH_OUTPUT

instead of wiki recommendation: https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/issues/92#issuecomment-430783036

jneilliii, 2023-05-26:
Updated examples on the rc branch in commit https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/commit/64233d3e0b55d766e10f30e7a6981a9b6aeb11c8, will go live on next update.

closed 2023-06-18 by github-actions[bot]

TheBoroer, 2024-04-12:
Also can confirm this works:

```
BED_MESH_CALIBRATE
BED_MESH_OUTPUT
```

and the klipper instructions in the wiki only outputs the last mesh data (doesn't actually probe the bed)

jneilliii, 2024-04-12:
Thanks for verifying. The latest rc still hasn't been released yet stable, but those are in the examples once the next version does go live. 

https://github.com/jneilliii/OctoPrint-BedLevelVisualizer/blob/rc/wiki/gcode-examples.md#klipper