jneilliii/OctoPrint-BedLevelVisualizer #617: [FR]: Report and record gantry level accuracy

RatchetHamster, 2023-06-01:
**Is your feature request related to a problem? Please describe.**

No, no bugs

**Describe the solution you'd like**

A simple value box that can extract and record the last known accuracy of a G34 iteration (gantry level for dual z steppers, independently driven)

**Describe alternatives you've considered**

Scroll through the terminal

**Additional context**

I do a lot of small prints and I don't bed level with each print, I just do detailed ones periodically. Before G29, I do a G34 and it would be really cool to have all the info of the accuracy achieved in one place if possible.

jneilliii, 2023-06-01:
Since I don't have a second z stepper driver to control individually I don't have G34 support so don't know how the output is. How would you invision this information to be visualized?

RatchetHamster, 2023-06-01:
Was thinking either just a box with a single value for current accuracy or
a convergence graph. Ah ok will be hard to test, I will see if I can get
hold of the source code and have a bash at putting something together.

jneilliii, 2023-06-01:
I can easily handle the coding side, because the plugin is a mess supporting so many variants of mesh reports from different firmware. I just needed a rough idea of how it should look and where in the current layout as well as how the commands respond in the terminal tab when you send them.

RatchetHamster, 2023-06-03:
Fair enough.

I have attached a file called "G34 I8.txt" which is the terminal output
when I send "G34 I8" to the printer. Note: my accuracy is 0.03mm, and if it
achieves that accuracy early it does not do all iterations.

If you are adding a box - another suggestion for a box alongside it to
include Probe Repeatability check. I have attached "G48 P5 X120 Y110.txt"
as well for the output from that - but I am sure you have a probe anyway.

Obviously both of these values are just "last known values", but cool
information to have to hand and check out.

The whole plugin is awesome by the way! Custom commands it a great to have
as well!

Kind Regards,

Janner

Send: M48 P5 X120 Y110
Recv: M48 Z-Probe Repeatability Test
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.34 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.34 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.34 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.34 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.50 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
[... 94 lines cut ...]
Recv: 
Recv: Z2-Z1=0.02
Recv: Target accuracy achieved.
Recv: Did 2 of 8
Recv: Accuracy: 0.02
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.97 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.81 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.81 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.81 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0
Recv: echo:busy: processing
Recv:  T:22.50 /0.00 B:22.81 /0.00 @:0 B@:0
Recv: X:153.00 Y:119.00 Z:11.20 E:0.00 Count X:12272 Y:9545 Z:4526
Recv: ok
Send: M113 S2
Recv: ok

jneilliii, 2023-06-03:
> Recv: Target accuracy achieved. 
Recv: Did 2 of 8 
Recv: Accuracy: 0.02 
Recv: echo:busy: processing 
Recv: T:22.66 /0.00 B:22.97 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.50 /0.00 B:22.66 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.66 /0.00 B:22.81 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.66 /0.00 B:22.81 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.50 /0.00 B:22.81 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.66 /0.00 B:22.66 /0.00 @:0 B@:0 
Recv: echo:busy: processing 
Recv: T:22.50 /0.00 B:22.81 /0.00 @:0 B@:0 
Recv: X:153.00 Y:119.00 Z:11.20 E:0.00 Count X:12272 Y:9545 Z:4526 
Recv: ok

I find this bit interesting at the end. What is the printer doing during this last bit of "busy: processing" until the `ok` is received?

RatchetHamster, 2023-06-05:
So it does the Gantry level where it moves left to right. Then once the
accuracy is achieved, I have "#define HOME_AFTER_G34" uncommented in my
marlin, so it homes all axis. Just after it has finished homing it gives
the " Recv: X:153.00 Y:119.00 Z:11.20 E:0.00 Count X:12272 Y:9545 Z:4526"
line, then "ok" followed by the "M113 S2" and another "ok". So I think
these commands are issued when the G34 has finished processing.

Kind Regards,

Janner