jneilliii/OctoPrint-BedLevelVisualizer #620: Action upon a measurement exceeding a threshold + or -

AlexVercammen, 2023-06-26:
I would love for Bed Visualizer to Cancel/Continue the start of a print if a measurement is exceeding a threshold + or - (user settings).
In my case, every prints start with the command to level with @BEDVISUALIZER. But, sometimes, a corner is out-of-wack and I have to catch it. Not sure if you can "activate" the Bed Visualizer tab to bring it upfront as you prompt the message. 
<<
M117 Homing...
G92 E0 ;Reset Extruder
G28 ;Home all axes
M117 Building Mesh...
G29 ;Level bed (mesh)
@BEDLEVELVISUALIZER
M420 S1 V ;Mesh Leveling ENABLED
>>

jneilliii, 2023-06-26:
> Not sure if you can "activate" the Bed Visualizer tab to bring it upfront as you prompt the message.

This would definitely be possible, but would probably make it an option. 

I think I've had a request before about the variance exceeding a threshold, do you mean that or each probe point?

AlexVercammen, 2023-06-27:
I think for each point. And if 1 of them exceeds the threshold prompt to Continue or Cancel. But the variance could work as well. Also, is 1 value for the threshold enough (Delta would **_not_** be signed) or do we need positive and negative values?
I think the latter would provide more flexibility. 
For example, I do flip my Glass Borosilicate bed depending if I use glue (glass side) or hairspray (Borosilicate side). And I've noticed the leveling results are different between the 2. Meaning, glass has to be warped at the corners. But I cannot tell you if the Delta is the same for both sides.

jneilliii, 2023-06-27:
in theory the delta should be the same from one side to the other. the variance value is exactly that, the difference between the highest point and the lowest point of all probed values.

AlexVercammen, 2023-07-08:
Do you think you can fix as well this issue of not rounding to 2 digit decimals for the variation calculation.

[image]

AlexVercammen, 2023-08-11:
Sorry about that. Thank you for entering it as an Enhancement.
Secondly, any update on this enhancement. I know there was a major release which may have impeded your progress.

jneilliii, 2023-08-11:
I work on plugins in my spare time and that has been minimal lately.